qmk/qmk_firmware #17946: RGB_ constants block RGB_VAD and RGB_VAI

max3-2, 2022-08-08:
Hi,

I am using a HSV_ORANGE and other color constants for setup and lighting layers, like so

in _config.h_
`#define RGBLIGHT_STARTUP_MODE	HSV_ORANGE`

or in _keymap.c_ following the layer lighting tutorial
```
const rgblight_segment_t PROGMEM my_win_fn_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {0, 8, HSV_CYAN}
);
```

This works very good with the single exception that after the call to a `HSV_` constant any following `RGB_VAI` or  `RGB_VAD` seem to have no effect.

Is this a bug? Is there a way to get both working? I have no issues setting the HSVs myself but I cant seem to find a good funtion that works in `RGBLIGHT_LAYER_SEGMENTS(...);`

drashna, 2022-08-14:
Do you mean that you want to change the HSV value later on?

max3-2, 2022-08-14:
Yes, I want to change the value (e.g. brightness) as one would do using RGB_VAD and RGB_VAI but thats blocked by the layer settings.

drashna, 2022-08-14:
I think you want `#define RGBLIGHT_LAYERS_RETAIN_VAL` then.  That looks to use the boards current val, rather than the hard set one.

max3-2, 2022-08-15:
Yes, perfect, thank you. I just skipped that little part in the documentation no idea why. Thanks!

closed 2022-08-15 by max3-2