qmk/qmk_firmware #17960: The layer cannot exceed 17 layers. If the layer exceeds 17 layers, some layer functions will have errors.

jeromeli002, 2022-08-09:
The layer cannot exceed 17 layers. If the layer exceeds 17 layers, some layer functions will have errors. For example, when qmk has 20 layers, when TO (0) is set, the output is TO (16). Is this a qmk problem or a vial problem?

tzarc, 2022-08-09:
Is this still a problem when you add `#define LAYER_STATE_32BIT` to `config.h`?

jeromeli002, 2022-08-09:
> Is this still a problem when you add `#define LAYER_STATE_32BIT` to `config.h`?

OK, thank you

jeromeli002, 2022-08-09:
> Is this still a problem when you add `#define LAYER_STATE_32BIT` to `config.h`?

Invalid, still the same situation,

zvecr, 2022-08-09:
<https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum_keycodes.h#L792>

> 16 layers max

Not a problem, its by design.

zvecr, 2022-08-14:
17989

closed 2022-08-14 by zvecr