qmk/qmk_firmware #17965: Is the link to the RGB Matrix Subsystem in the RGB matrix doc page broken?

pitommasi, 2022-08-09:
I'm referring to this page:
https://github.com/qmk/qmk_firmware/blob/master/docs/feature_rgb_matrix.md
 
in the first paragraph it says:
"If you want to use RGB LED's you should use the [RGB Matrix Subsystem](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_rgb_matrix.md) instead."

as you can see, the link points to the same page, where I can't find any reference to thuis RGB Matrix subsystem...
Bear with me, I'm new to QMK so maybe I am the problem...

waffle87, 2022-08-09:
Unsure where you're seeing that. The current doc in master points to the LED matrix page.
> If you want to use single color LED's you should use the [LED Matrix Subsystem](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_led_matrix.md) instead.

drashna, 2022-08-14:
May be a caching issue from an older revision?  Hard refreshing may fix the issue.

zvecr, 2022-08-15:
.md files are fine

closed 2022-08-15 by zvecr