qmk/qmk_firmware #17977: [Bug] Keymap sofle/rev1:devdev won't compile. Firmware to large.

duese, 2022-08-10:
## Describe the Bug

I tried to compile the keymap "devdev" for the RGB variant of the sofle keyboard using the following command:

```
qmk compile -kb sofle/rev1 -km devdev
```

I got the following compilation error:
```text
Ψ Compiling keymap with gmake --jobs=1 sofle/rev1:devdev

QMK Firmware 0.17.7
Making sofle/rev1 with keymap devdev

avr-gcc (Homebrew AVR GCC 8.5.0) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_sofle_rev1/src/info_config.h                                                 [OK]
Generating: .build/obj_sofle_rev1/src/default_keyboard.c                                            [OK]
Generating: .build/obj_sofle_rev1/src/default_keyboard.h                                            [OK]
Generating: .build/obj_sofle_rev1/src/layouts.h                                                     [OK]
Compiling: keyboards/sofle/sofle.c                                                                  [OK]
Compiling: keyboards/sofle/rev1/rev1.c                                                              [OK]
Compiling: .build/obj_sofle_rev1/src/default_keyboard.c                                             [OK]
Compiling: keyboards/sofle/keymaps/devdev/keymap.c                                                  [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/send_string.c                                                                    [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/split_common/split_util.c                                                        [OK]
Compiling: quantum/split_common/transport.c                                                         [OK]
Compiling: quantum/split_common/transactions.c                                                      [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: quantum/mousekey.c                                                                       [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight/rgblight.c                                                              [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: platforms/avr/drivers/ws2812.c                                                           [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/crc.c                                                                            [OK]
Compiling: drivers/oled/ssd1306_sh1106.c                                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/encoder.c                                                                        [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: platforms/avr/drivers/i2c_master.c                                                       [OK]
Archiving: .build/obj_sofle_rev1_devdev/i2c_master.o                                                [OK]
Compiling: platforms/avr/drivers/i2c_slave.c                                                        [OK]
Archiving: .build/obj_sofle_rev1_devdev/i2c_slave.o                                                 [OK]
Compiling: platforms/avr/drivers/serial.c                                                           [OK]
Archiving: .build/obj_sofle_rev1_devdev/serial.o                                                    [OK]
Assembling: platforms/avr/xprintf.S                                                                 [OK]
[... 53 lines cut ...]
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

waffle87, 2022-08-11:
[Squeezing AVR](https://docs.qmk.fm/#/squeezing_avr)

tzarc, 2022-08-29:
Under normal circumstances it's up to the keymap maintainer to ensure compilation is successful -- QMK will keep a handful of things up to date like changed keycodes... but in general if things don't build we don't concern ourselves with it.

closed 2022-08-29 by tzarc