qmk/qmk_firmware #17978: [Bug] Compilation of keymap sofle:flare576 fails. Unused function.

duese, 2022-08-10:
## Describe the Bug

I tried to compile the keymap sofle:flare576 using the following command:

```
qmk compile -kb sofle -km flare576
```

The compilation failed:
```text
Ψ Compiling keymap with gmake --jobs=1 sofle/rev1:flare576

QMK Firmware 0.17.7
Making sofle/rev1 with keymap flare576

avr-gcc (Homebrew AVR GCC 8.5.0) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_sofle_rev1/src/info_config.h                                                 [OK]
Generating: .build/obj_sofle_rev1/src/default_keyboard.c                                            [OK]
Generating: .build/obj_sofle_rev1/src/default_keyboard.h                                            [OK]
Generating: .build/obj_sofle_rev1/src/layouts.h                                                     [OK]
Compiling: keyboards/sofle/keymaps/flare576/graphics.c                                             keyboards/sofle/keymaps/flare576/graphics.c:288:13: error: 'luna_jump' defined but not used [-Werror=unused-function]
 static void luna_jump(bool newJump){}
             ^~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:456: .build/obj_sofle_rev1_flare576/graphics.o] Error 1
gmake: *** [Makefile:413: sofle/rev1:flare576] Error 1
Make finished with errors
```

## System Information

**Keyboard:** Sofle
**Revision (if applicable):** 
**Operating system:** OSX 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /Users/duese/dev/experiments/sofle/qmk
Ψ Detected macOS 12.4.
Ψ Git branch: master
Ψ Repo version: 0.17.7
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

waffle87, 2022-08-11:
Unsure if this will produce expected results on hardware but
```diff
diff --git a/keyboards/sofle/keymaps/flare576/graphics.c b/keyboards/sofle/keymaps/flare576/graphics.c
index 918334a508..e2f88afe26 100644
--- a/keyboards/sofle/keymaps/flare576/graphics.c
+++ b/keyboards/sofle/keymaps/flare576/graphics.c
@@ -229,7 +229,7 @@ static void main_board(void) {
 // End "MAIN_BOARD"
 #else
 static void main_board(void){}
-static void luna_jump(bool newJump){}
+void luna_jump(bool newJump){}

 #define MARQUEE_FRAME_DURATION 200 // how long each frame lasts in ms
 #define LOGO_BYTES 128
@@ -285,6 +285,6 @@ void oled_task_user(void) {

 // End "OLED_DRIVER_ENABLE"
 #else
-static void luna_jump(bool newJump){}
+void luna_jump(bool newJump){}
 // End "OLED_DRIVER_ENABLE" false
 #endif```

tzarc, 2022-08-29:
Under normal circumstances it's up to the keymap maintainer to ensure compilation is successful -- QMK will keep a handful of things up to date like changed keycodes... but in general if things don't build we don't concern ourselves with it.

closed 2022-08-29 by tzarc