qmk/qmk_firmware #17986: YMDK YM68 ISO layout [Feature Request]

marcoXbresciani, 2022-08-10:
Hello,
  I'm connecting to #12339 and #12352 as reference.

I'm "working with" YMDK support to have a custom YM68 with Italian ISO layout and I was wondering if it would be possible to add support for the ISO layout as well.
Have no idea myself how to do that, but I could help with keys position and the like.

Thanks

noroadsleft, 2023-07-17:
@marcoXbresciani,

Sorry for the slow response. I've since added ISO layout support for the YM68 in https://github.com/qmk/qmk_firmware/pull/20906, but I never got confirmation as to whether it worked properly; I made a semi-educated guess. If you can test it and let me know if it works properly, that would be great.

marcoXbresciani, 2023-07-17:
Thank you @noroadsleft.
Unfortunately I can't test it anymore: in order to have better posture and less pains here and there, I've been suggested to move to an ergo split keyboard, so I don't have it anymore.

noroadsleft, 2023-07-17:
Thanks for the response.

Sorry to hear of your health troubles; best wishes moving forward. Hopefully you find something that suits your needs and preferences.

closed 2023-07-17 by noroadsleft