qmk/qmk_firmware #17993: [Bug] QK_BOOT or reset_keyboard() does not restart to bootloader

npoirey, 2022-08-11:
## Describe the Bug

QK_BOOT or reset_keyboard() does not restart to bootloader, but instead seems to soft reboot and have the same effect than QK_RBT
I have to press the button on the teensy to enter bootloader for now.

## System Information

**Keyboard:** https://github.com/npoirey/qmk_firmware/tree/master/keyboards/handwired/frenchdev/monoblock
**Revision (if applicable):** 
**Operating system:** windows
**`qmk doctor` output:** 
[Sacapuces@DESKTOP-S8V260T qmk]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: D:/Repositories/qmk
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: master
Ψ Repo version: 0.17.7
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ QMK is ready to go, but minor problems were found

**Any keyboard related software installed?** 
 - [x] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context
built after merging with last qmk master

sigprof, 2022-08-11:
Looks like your `info.json` contains `"bootloader": "atmel-dfu"`; if you actually use a Teensy board, the bootloader type must be set to `halfkay`.

npoirey, 2022-08-11:
Yep, that was it. Thank you for the help ! I must have missed something when using the CLI to create a second version.

closed 2022-08-11 by npoirey

leep-frog, 2024-09-11:
I'm having a similar problem with my arduino leonardo.  I currently have `"bootloader": "atmel-dfu",` in my info.json.  Is that correct?

fauxpark, 2024-09-11:
Arduino boards use Caterina.