qmk/qmk_firmware #17995: [Bug] Installation via Homebrew fails on M1 Mac

thought-tobi, 2022-08-11:
The installation of qmk via Homebrew on my MacBook seems to fail because of the Pillow dependency.

## Describe the Bug

When running ```brew install qmk/qmk/qmk``` as outlined in the docs, the following happens:
 ```
==> Installing qmk from qmk/qmk
==> python3 -m venv --system-site-packages /opt/homebrew/Cellar/qmk/1.1.0/libexec
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--appdirs-20220811-67304-oeq65l/appdirs-1.4.4
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--argcomplete-20220811-67304-13ej6u7/argcomplete-2.0.0
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--attrs-20220811-67304-18gu0xm/attrs-21.4.0
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--colorama-20220811-67304-v04y8m/colorama-0.4.4
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--halo-20220811-67304-msdi49/halo-0.0.31
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--hid-20220811-67304-se1i7t/hid-1.0.5
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--hjson-20220811-67304-nq8qem/hjson-3.0.2
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--jsonschema-20220811-67304-1dsmjqa/jsonschema-4.5.1
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--log-symbols-20220811-67304-1rm25mw/log_symbols-0.0.14
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--milc-20220811-67304-voub9x/milc-1.6.6
==> /opt/homebrew/Cellar/qmk/1.1.0/libexec/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/qmk--Pillow-20220811-67304-kv9nbp/Pillow-9.1.1
Last 15 lines from /Users/tobi/Library/Logs/Homebrew/qmk/12.pip:

  exec(compile(setup_py_code, filename, "exec"))
  '"'"''"'"''"'"' % ('"'"'/private/tmp/qmk--Pillow-20220811-67304-kv9nbp/Pillow-9.1.1/setup.py'"'"',), "<pip-setuptools-caller>", "exec"))' install --record /private/tmp/pip-record-40hmo2gp/install-record.txt --single-version-externally-managed --compile --install-headers /opt/homebrew/Cellar/qmk/1.1.0/libexec/include/site/python3.9/Pillow
  cwd: /private/tmp/qmk--Pillow-20220811-67304-kv9nbp/Pillow-9.1.1/
  Running setup.py install for Pillow: finished with status 'error'
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> Pillow

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

I have tried the following (to no success):

- ```python3 -m pip install --upgrade pip```
- ```python3 -m pip install --upgrade Pillow```
- Followed the steps outlined in [this thread](https://github.com/python-pillow/Pillow/issues/5093), manually downloading, renaming and installing the corresponding wheel
- running ```brew cleanup -s``` to make sure there were no issues with cached packages or anything like that
- Running the commands from a fresh venv to make sure that the issue isn't with my bloated global python environment

## System Information

**Operating system:** macOS Monterey 12.2.1
**Homebrew version:** Homebrew 3.5.9
**Python version:** Tested with both 3.8 and 3.9, both unsuccessfully

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

fauxpark, 2022-08-11:
Installing the CLI this way won't work, you need to install Homebrew with Rosetta 2 and install the CLI there.

https://www.reddit.com/r/olkb/comments/nh2fk9/guide_installing_qmk_on_m1_macbook/

closed 2022-08-11 by fauxpark

burlapsax, 2022-08-21:
@fauxpark , I am experiencing precisely the same behavior reported by @thought-tobi , but I am using an Intel Mac, which is running macOS 10.14.6, and which has successfully compiled qmk in the past without this problem. I believe this issue should be reopened.

fauxpark, 2022-08-21:
That is a different problem.

Homebrew no longer supports Mojave: https://docs.brew.sh/Installation#macos-requirements
Additionally, Github Actions only provides runner images for 10.15 and above (and even that appears to be deprecated now): https://github.com/actions/runner-images
So, bottles are only built for Catalina; below that, brew will attempt to build the formula from source, which is not necessarily guaranteed to work.

AlchemR, 2024-04-15:
> @fauxpark , I am experiencing precisely the same behavior reported by @thought-tobi , but I am using an Intel Mac, which is running macOS 10.14.6, and which has successfully compiled qmk in the past without this problem. I believe this issue should be reopened.

Did you ever get a decent answer?  I'm in the same issue and I couldn't see any follow up.

fauxpark, 2024-04-15:
As I mentioned above, Homebrew no longer supports 10.14; in fact the minimum macOS version as of writing is Monterey. As far as Apple, Homebrew and we are concerned you are very much out of support.