qmk/qmk_firmware #18001: [Bug] Discrepancies between `qmk compile keymap.json` and `qmk compile -kb … -km …`

precondition, 2022-08-11:
## Describe the Bug

When tasked to compile the exact same keymap.json file, `qmk compile keymap.json` and `qmk compile -kb ez_maker/directpins/teensy_lc -km default` (or running it while in the keymaps directory) do not compile the same firmware as evidenced by the fact that the steno feature is correctly compiled in one case but is ignored in the other case.

This issue is not limited to the DirectPins Teensy LC keyboard, that's just an example.

```zsh
~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8
❯ qmk clean -a   
QMK Firmware 0.17.7
Deleting .build/ ... done.
Deleting *.bin, *.hex, and *.uf2 ... done.

~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8                                      
❯ ls
keymap.json

~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8                                         
❯ cat keymap.json 
{
    "config": {
        "stenography": {
            "enabled": true
        }
    },
    "keyboard": "ez_maker/directpins/teensy_lc",
    "keymap": "default",
    "layout": "LAYOUT_all",
    "layers": [
        [
            "KC_0",
            "KC_1",
            "KC_2",                 "KC_3",
            "KC_4",         "KC_5", "KC_6",
            "KC_7",         "KC_8", "KC_9",
            "KC_A",                 "KC_B",
            "KC_C",                 "KC_D",
            "KC_E",                 "KC_F",
            "KC_G",                 "KC_H",
            "KC_I",                 "KC_J",
            "KC_K",                 "KC_L",
            "KC_M",                 "KC_N",
            "KC_O",         "KC_P", "KC_Q"
        ]
    ]
}

~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8                                         
❯ qmk compile 2>&1 | grep steno
Compiling: quantum/process_keycode/process_steno.c                                                 Compiling: lib/printf/src/printf/printf.c                                                           [OK]

~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8                                 took 12s
❯ qmk clean -a
QMK Firmware 0.17.7
Deleting .build/ ... done.
Deleting *.bin, *.hex, and *.uf2 ... done.

~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8                                         
❯ qmk compile keymap.json 2>&1 | grep steno

~/qmk_firmware/keyboards/ez_maker/directpins/teensy_lc/keymaps/default on develop *5 !1 ⨤8                            ✘ 0|1 took 8s
❯ 

```

## System Information

**Keyboard:** Irrelevant
**Revision (if applicable):** Irrelevant
**Operating system:** Xubuntu
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/vern/qmk_firmware
Ψ Detected Linux.
⚠ Found old, deprecated udev rules for 'atmel-dfu' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Found old, deprecated udev rules for 'kiibohd' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Found old, deprecated udev rules for 'stm32' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Found old, deprecated udev rules for 'bootloadhid' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/vern/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'massdrop' boards. Run 'sudo cp /home/vern/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Found old, deprecated udev rules for 'caterina' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/vern/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: develop
Ψ Repo version: 0.17.7
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest develop: 2022-08-11 13:28:48 -0400 (272f224dc6) -- Add cursor layer to DMQ Spin (#17996)
Ψ - Latest upstream/master: 2022-08-12 00:08:57 +0800 (55f5406ac7) -- update ecila layout (#17975)
Ψ - Latest upstream/develop: 2022-08-11 21:04:46 +0000 (8c4269f98e) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2022-08-12 00:08:57 +0800 (55f5406ac7) -- update ecila layout (#17975)
Ψ - Common ancestor with upstream/develop: 2022-08-11 13:28:48 -0400 (272f224dc6) -- Add cursor layer to DMQ Spin (#17996)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 09:32:15 +0200 --  (966c4889)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-06-12 22:00:28 +1000 --  (35cc3d92f)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

zvecr, 2022-08-11:
```diff
index 9c0a5dce56..afe0acc50b 100644
--- a/lib/python/qmk/commands.py
+++ b/lib/python/qmk/commands.py
@@ -3,12 +3,12 @@
 import os
 import sys
 import shutil
+import json
 from pathlib import Path
 
 from milc import cli
 import jsonschema
 
-import qmk.keymap
 from qmk.constants import KEYBOARD_OUTPUT_PREFIX
 from qmk.json_schema import json_load, validate
 
@@ -134,12 +134,11 @@ def compile_configurator_json(user_keymap, bootloader=None, parallel=1, **env_va
     target = f'{keyboard_filesafe}_{user_keymap["keymap"]}'
     keyboard_output = Path(f'{KEYBOARD_OUTPUT_PREFIX}{keyboard_filesafe}')
     keymap_output = Path(f'{keyboard_output}_{user_keymap["keymap"]}')
-    c_text = qmk.keymap.generate_c(user_keymap)
     keymap_dir = keymap_output / 'src'
-    keymap_c = keymap_dir / 'keymap.c'
+    keymap_json = keymap_dir / 'keymap.json'
 
     keymap_dir.mkdir(exist_ok=True, parents=True)
-    keymap_c.write_text(c_text)
+    keymap_json.write_text(json.dumps(user_keymap))
 
     # Return a command that can be run to make the keymap and flash if given
     verbose = 'true' if cli.config.general.verbose else 'false'
@@ -175,7 +174,7 @@ def compile_configurator_json(user_keymap, bootloader=None, parallel=1, **env_va
         f'MAIN_KEYMAP_PATH_3={keymap_output}',
         f'MAIN_KEYMAP_PATH_4={keymap_output}',
         f'MAIN_KEYMAP_PATH_5={keymap_output}',
-        f'KEYMAP_C={keymap_c}',
+        f'KEYMAP_JSON={keymap_json}',
         f'KEYMAP_PATH={keymap_dir}',
         f'VERBOSE={verbose}',
         f'COLOR={color}',

```

precondition, 2022-10-24:
Fixed by #18766

closed 2022-10-24 by precondition