qmk/qmk_firmware #18077: [Bug] PS/2 interrupt handler silently drops bytes in case of errors

joh, 2022-08-16:
## Describe the Bug

The PS/2 interrupt handler (`ps2_interrupt_service_routine`) places incoming bytes on a ring buffer, for later processing by the relevant driver (e.g. `ps2_mouse_task`).
In case there is an error in the incoming data (e.g. parity check failure), the interrupt handler will silently drop the received byte, and set a global `ps2_error` flag.
For the downstream driver (`ps2_mouse`), it is impossible to recover from an error, since it requires a 3-byte packet of movement data.
Consequently, the driver receives three bytes which actually belong to different 3-byte packets!

Consider the following stream of incoming PS/2 bytes, where the letters indicate which 3-byte packet each byte belongs to:

`A1 A2 A3 B1 B2 B3 C1 C2 C3`

Imagine byte `A2` contains a parity error, causing it to be silently dropped by the interrupt handler.
The resulting ring buffer will then contain:

`A1 A3 B1 B2 B3 C1 C2 C3`

The `ps2_driver`, unknowingly consumes `A1` `A3` and `B1` as a 3-byte movement packet, resulting in very strange mouse movements.
To make matters worse, all the packets following the error will also be out of sync (`B2 B3 C1`).

## Some background

I have a trackpoint module which, for some reason, doesn't respond correctly to the Read Data command.
However, in stream mode, it will send movement packets continously whenever any movement is detected.
As far as I understand (and please correct me if I'm wrong), this is the correct behavior for stream mode.

> Stream Mode:
> 
> In stream mode, the mouse sends movement data when it detects movement or a change in state of one or more mouse buttons. The maximum rate at which this data reporting may occur is known as the sample rate.  This parameter ranges from 10 samples/sec to 200 samples/sec. Its default value is 100 samples/sec and the host may change that value by using the "Set Sample Rate" (0xF3) command.  Stream mode is the default mode of operation.

(From https://web.archive.org/web/20041117095622/http://panda.cs.ndsu.nodak.edu:80/~achapwes/PICmicro/mouse/mouse.html )

`ps2_mouse_task` always issues a Read Data command, regardless of mode, before reading a 3-byte movement packet.
I have modified `ps2_mouse_task` to not send the Read Data command, and simply process the data as it lands in the ring buffer.
For some weird reason, the parity check fails every now and again, causing the input stream to become out of sync, as described above.
I'm not sure if this issue exists for other PS/2 devices that do respond correctly to the Read Data command, but if they also send movement packets continuously in stream mode, I suspect it will.

## Possible solution:

For the `ps2_mouse` driver, I think it is safe to simply ignore any 3-byte packet with errors.
To do so, the mouse driver needs to know if any of the 3 bytes contain errors.
Hence the interrupt handler must stop dropping bytes, and each byte should be accompanied with an error flag.
For example:
1. Change data type of ring buffer to `uint16_t`
2. Use lower byte for data, upper byte for error flags
3. Modify `ps2_host_recv` and `ps2_host_recv_response` to return an error code (0 for success), and a pointer to where the data should be written.

Alternatively, one could try issuing a Resend command, but this is probably overkill.

I would be happy to implement something like this and submit a PR, but would first like to hear your thoughts. Perhaps it's only a problem with my peculiar PS/2 device :)

## System Information

**Keyboard: crkbd** 
**Revision (if applicable): v3** 
**Operating system: Linux** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/joh/code/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.17.6
Ψ - Latest master: 2022-08-03 12:23:17 -0500 (df8a538489) -- Userspace: muppetjones (#1) (#13461)
Ψ - Latest upstream/master: 2022-08-16 20:41:31 +0200 (837a57b3c5) -- Adjust the hierarchy of chapter `Deferred Execution` (#18075)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-08-03 12:23:17 -0500 (df8a538489) -- Userspace: muppetjones (#1) (#13461)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

KarlK90, 2022-08-17:
Thanks for the bug report. The error handling is definitely lacking in the PS2 driver and from looking at the code I think we could keep most of the existing design and do the following:

1. `ps2_mouse_task` sets the global `ps2_error` at the beginning to `PS2_ERR_NONE`, all PS2 receive/transmit functions of the drivers no longer clear this error flag, but only add their error bits instead eg.  `ps2_error |= PS2_ERR_PARITY`. In this way any error condition that happens in a failed transfer is not overwritten by a subsequent successful transfer.
2. An error check is added to https://github.com/qmk/qmk_firmware/blob/develop/drivers/ps2/ps2_mouse.c#L108 and the branch is only taken if `ps2_error == PS2_ERR_NONE`, so if there have been any errors in the transfer we discard all of it.
3. (Optional?) Do cleanup on error e.g. flush any buffers that contain half-finished transfers

If I understand your explanations correctly this should keep the system in sync and discard any malformed data with the one downside that all data of a transfer is discarded, even "good bytes".

joh, 2022-08-17:
Thank you for the quick reply :)

I don't see how your proposed modifications would solve the sync issue caused by the dropping of any bytes with errors though. Consider the following stream of 3-byte packets:

`A1 A2 A3 B1 B2 B3 C1 C2 C3`

Let's assume `A2` contains a parity error. If I understand your suggestion correctly, this would cause the interrupt handler (?) to add PS2_ERR_PARITY to `ps2_error`, but still drop the `A2` byte. The resulting ring buffer would then contain:

`A1 A3 B1 B2 B3 C1 C2 C3`

Next, `ps2_mouse_task` will consume `A1 A3 B1` from the ring buffer. However, it will notice that `ps2_error != PS2_ERR_NONE` and discard that packet. It will then clear the `ps2_error` flag(?)

The ring buffer now contains:

`B2 B3 C1 C2 C3`

Next iteration `ps2_mouse_task` is called again, and it will consume `B2 B3 C1`. This time around, there is no error (`ps2_error == PS2_ERR_NONE`), and it will happily use this malformed packet. To be clear, the problem is that `B2` will be interpreted as the first byte in a packet (expected to contain button status, sign bits and overflow bits), while `B2` in reality contains X Movement data.

Another alternative could be to request the PS2 device to resend the packet:

> The host may inhibit communication at any time by pulling the Clock line low for at least 100 microseconds.  If a transmission is inhibited before the 11th clock pulse, the device must abort the current transmission and prepare to retransmit the current "chunk" of data when host releases Clock.  A "chunk" of data could be a make code, break code, device ID, mouse movement packet, etc.  For example, if a keyboard is interrupted while sending the second byte of a two-byte break code, it will need to retransmit both bytes of that break code, not just the one that was interrupted.
(See http://www.burtonsys.com/ps2_chapweske.htm)

Alternatively, by sending the 0xFE (Resend) command.

In any case, whenever there is an error detected, the ring buffer must be cleared before requesting the resend, in order to bring the byte order back in sync.

joh, 2022-09-27:
FWIW, I have found that the parity errors are due to [delayed execution of the PS/2 interrupt handler](https://www.reddit.com/r/olkb/comments/xppdsz/ps2_interrupt_delays_on_avr/) `ps2_interrupt_service_routine`, when other parts of QMK disable interrupts for too long (such as soft serial).

sigprof, 2022-09-27:
Probably can't do anything about the soft serial, because that code has strict timing requirements.

If you are stuck with AVR, your options are:
- Use remote mode for PS/2 if it works with your device (although this still might not help if the PS/2 device is on the slave side, because in that case the soft serial interrupt may come asynchronously during the PS/2 data transfer, and apparently the PS/2 code does not disable interrupts).
- Use I2C for split communications, which does not disable interrupts on the master side of the split (but the slave side might still have some problems when the slave callback gets called from the I2C interrupt handler).
- Try using the USART mode for PS/2, which might have less strict interrupt latency requirements (unfortunately, that mode requires pins which are not broken out on a Pro Micro).

50an6xy06r6n, 2023-12-11:
Any movement on this? I'm seeing a similar issue with my trackpoint (though mine works in remote mode). I tried implementing @KarlK90's solution and it helped a bit, but I still occasionally see instances of the packets getting out of sync by one byte, which results in erratic movement and mouse clicks.

Is this just kind of unfixable given the execution delay issue?

joh, 2023-12-12:
> Any movement on this? I'm seeing a similar issue with my trackpoint (though mine works in remote mode). I tried implementing @KarlK90's solution and it helped a bit, but I still occasionally see [...]

The PS/2 code is really not fault tolerant, so your best bet is to fix the underlying issue such that you don't get any PS/2 errors in the first place. If possible, consider switching to the USART PS/2 driver (for AVR) or the PIO PS/2 driver (for RP2040). I'm using the latter and experience no errors.