qmk/qmk_firmware #18089: [Doc] re-organization of the chapter `How to Customize Your Keyboard’s Behavior`

coliss86, 2022-08-17:
Hi, 

I think that the chapters [`Layer Change Code`](https://docs.qmk.fm/#/custom_quantum_functions?id=layer-change-code) and [`Persistent Configuration (EEPROM)`](https://docs.qmk.fm/#/custom_quantum_functions?id=persistent-configuration-eeprom) in the page [How to Customize Your Keyboard’s Behavior](https://docs.qmk.fm/#/custom_quantum_functions?id=how-to-customize-your-keyboards-behavior) should be moved to gain a better visibility, those pages are hard to reach.

I proposed to do the following:
- Move `Layer Change Code` at the end of the page [Layers](https://docs.qmk.fm/#/feature_layers?id=layers)
- Move `Persistent Configuration (EEPROM)` in its own page and referenced in the `Software Features` with the title `EEPROM`

WDYT ?

If you are OK, I'll do a PR.

precondition, 2022-08-19:
Sounds good to me.

coliss86, 2022-11-18:
Fixed in https://github.com/qmk/qmk_firmware/pull/18174

closed 2022-11-18 by coliss86