qmk/qmk_firmware #18094: [Bug] ps2_usart.o No such file or directory | PS/2 with USART version | STM32F401CCU6 WeAct BlackPill

delilcodes, 2022-08-18:
## Describe the Bug

I use an touchpad with keyboard so I tried before Interrupt version but nothing works but with BUSYWAIT version I can use the touchpad. I tried USART version but the compiler (qmk msys) has interrupt with that issue. How can i find this file. I searched the repo from qmk firmware but did not found anything about that file.

## System Information

**Keyboard:** 
**Revision (if applicable):** 
**Operating system:** 
**`qmk doctor` output:** 
```
qmk doctor output;

Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: C:/Users/degir/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: master
Ψ Repo version: 0.17.9
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ QMK is ready to go, but minor problems were found

```

## Additional Context

delilcodes, 2022-08-18:
https://i.imgur.com/crYzxDM.png

zvecr, 2022-08-18:
Not currently a supported config:

avr
<https://github.com/qmk/qmk_firmware/blob/master/platforms/avr/drivers/ps2/ps2_usart.c>

chibios
<https://github.com/qmk/qmk_firmware/tree/master/platforms/chibios/drivers/ps2> no `ps2_usart.c` file

closed 2022-12-18 by github-actions[bot]