qmk/qmk_firmware #18096: "Run failed at startup" Automatic Approve workflow run fails for pushes to private forks of qmk_firmware

jweickm, 2022-08-18:
Since today, when pushing to any branch of my private fork of this repo, I received emails informing me that a "Run failed at startup: Automatic Approve - master (#######)". 

Disabling actions on the settings page of my repo according to the [advice by [Mental_General_5445](https://www.reddit.com/user/Mental_General_5445/)](https://www.reddit.com/r/olkb/comments/wrlvd0/is_there_a_new_github_automatic_approve_workflow/) appears to suppress the email messages for now.

As this issue seems to affect at least one more person, @drashna suggested that there might be a way to filter out forks for automatic workflow actions for the repo.

zvecr, 2022-08-18:
Should already be filtering out forks:

https://github.com/qmk/qmk_firmware/blob/9607b46756ae24bd54f47449d575c8699701b057/.github/workflows/auto_approve.yml#L11

jweickm, 2022-08-18:
> Should already be filtering out forks:
> 
> https://github.com/qmk/qmk_firmware/blob/9607b46756ae24bd54f47449d575c8699701b057/.github/workflows/auto_approve.yml#L11

Could there be another reason then, why I suddenly received these emails? Never received anything before from pushing to my fork of this repo.

zvecr, 2022-08-18:
Link 404s now that actions have been disabled on the fork.

sigprof, 2022-08-18:
It might have been some internal GitHub problem — apparently `qmk/qmk_firmware` also had some failures of that workflow recently:
[image]
[image]
Although disabling that workflow is probably the right thing anyway, because otherwise it would be started every 5 minutes only to have its only job skipped; note that you can disable individual workflows if you want to use only some of them in your fork.

jweickm, 2022-08-18:
Thanks for all the responses. 
I guess we can close this issue now. If other people encounter a similar issue, maybe they'll find this thread to see that disabling the action workflow under "Settings -> Actions" will suppress any unintentional workflows for forks of this repo.

closed 2022-08-18 by jweickm

zvecr, 2022-08-18:
<https://www.githubstatus.com/incidents/8gpymn7rystj> seems there was an issue at that time