qmk/qmk_firmware #18105: QMK failed compiling self made keyboard

ottob88, 2022-08-19:
Hello everyone,

With QMK I've made in the past a self made keyboard work with a Proton C chip. Recently, I tried, as far as I think, the same without succes. Every time I get the following error while compiling. In attachments I've added the files I work with.

Please let me know, I really appreciate it!
[Archive.zip](https://github.com/qmk/qmk_firmware/files/9382666/Archive.zip)

QMK Firmware 0.17.9
⚠ johan: johan: Falling back to searching for KEYMAP/LAYOUT macros.
Making johan with keymap johan21

⚠ johan: johan: Falling back to searching for KEYMAP/LAYOUT macros.
Generating: .build/obj_johan/src/default_keyboard.c                                                 [WARNINGS]
 | 
 | ⚠ johan: johan: Falling back to searching for KEYMAP/LAYOUT macros.
 | 
avr-gcc (Homebrew AVR GCC 8.5.0) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_johan/src/info_config.h                                                      [WARNINGS]
 | 
 | ⚠ johan: johan: Falling back to searching for KEYMAP/LAYOUT macros.
 | 
Generating: .build/obj_johan/src/default_keyboard.h                                                 [WARNINGS]
 | 
 | ⚠ johan: johan: Falling back to searching for KEYMAP/LAYOUT macros.
 | 
Generating: .build/obj_johan/src/layouts.h                                                          [WARNINGS]
 | 
 | ⚠ johan: johan: Falling back to searching for KEYMAP/LAYOUT macros.
 | 
Compiling: .build/obj_johan/src/default_keyboard.c                                                 In file included from .build/obj_johan/src/default_keyboard.c:26:
./.build/obj_johan/src/default_keyboard.h:28:2: error: #error ("<keyboard>.h is only optional for data driven keyboards - kb.h == bad times")
 #error("<keyboard>.h is only optional for data driven keyboards - kb.h == bad times")
  ^~~~~
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:456: .build/obj_johan_johan21/.build/obj_johan/src/default_keyboard.o] Error 1
gmake: *** [Makefile:413: johan:johan21] Error 1
Make finished with errors

waffle87, 2022-08-19:
Looks like you used kbfirmware builder. That's extremely outdated and won't work with modern QMK — `qmk new-keyboard` would be a better place to start. But if you must, use [KBFirmware JSON to QMK Parser](https://noroadsleft.github.io/kbf_qmk_converter/) first. The error you're currently running into is because the `keyboard.c/h` file names do not match the parent directory name.

keyboard-magpie, 2022-08-29:
Closing this for now, if you need further support please join the QMK discord server.

closed 2022-08-29 by keyboard-magpie

paullemmens, 2023-01-11:
I recently ran into this weird error as well. Learned on discord that a `qmk clean -a` should do (and did) the trick. My error was `"<keyboard>.h is only optional for data driven keyboards`.

Noting this here for posterity.

BudujzJacula, 2023-05-04:
Hi, I run into the same error:
 `#error("<keyboard>.h is required unless your keyboard uses data-driven configuration. Please rename your keyboard's header file to <keyboard>.h")`
I started new project with (never touched KBfirmware):
`qmk new-keyboard`
I have a handwired keyboard, with double Pro Micro, custom wiring. Compiling on Windows 10, also tried on Ubuntu VM with the same result.
I tried using KBFirmawe JSON to QMK Parser but with no luck, tried `qmk clean -a` but also no luck.

Have You got any ideas what can I do to fix this issue?
Let me know if You need any additional info.

cupcake08, 2023-07-02:
Kindly check your info.json file again. After fixing the file , for example: matching the "matrix_pins" to "layout" correctly.

For me It resolved the issue.