qmk/qmk_firmware #18123: [Bug] `SEND_STRING` reorders argument input

leep-frog, 2022-08-21:
## Describe the Bug

```
SEND_STRING(SS_UP(X_RSFT) SS_RCTL("k"))
``` 

The above command sends `SS_UP(X_RSFT)` at the very end of the provided arguments, making it behave like the following: 

```
SEND_STRING(SS_RCTL("k") SS_UP(X_RSFT))
```

## System Information

**Keyboard:** ergodox_ez
**Revision (if applicable):** 
**Operating system:**: Windows
**`qmk doctor` output:** 
```
~ $ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.2.0
Ψ QMK home: C:/Users/gleep/qmk_firmware
Ψ Detected Windows 10 (10.0.22000).
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

I have the following code in my qmk configuration:

```
void TDShifter(qk_tap_dance_state_t *state, void *user_data) {
    if (state->count == 1) {
        SEND_STRING(SS_UP(X_RSFT) SS_RCTL("k"));
        return;
    }
    SEND_STRING(SS_UP(X_RSFT));
}

qk_tap_dance_action_t tap_dance_actions[] = {
    [TDK_SHIFTER]  = ACTION_TAP_DANCE_FN(TDShifter),
};

// I use CK_SHIFTER as a keycode in my keymap.c file
#define CK_SHIFTER TD(TDK_SHIFTER)
```

Double pressing the key sends the proper sequence (starts holding shift), but single pressing it sends the following (tested with [this website](https://config.qmk.fm/#/test/)):

| QMK key action | Key | Event key | Code | Key Code |
| --- | ---  | ---            | ---      | --- |
| Double tap CK_SHIFTER key | KEY-DOWN - QMK: KC_RSFT | Event key: Shift  |      Code: ShiftRight  |  KeyCode: 16 |
| Single tap CK_SHIFTER key | KEY-DOWN - QMK: KC_RCTL  | Event key: Control  |   Code: ControlRight |  KeyCode: 17 |
| | KEY-DOWN - QMK: KC_K  |  Event key: K       |    Code: KeyK    |      KeyCode: 75 |
| | KEY-UP   - QMK: KC_K  |  Event key: K       |    Code: KeyK |         KeyCode: 75 | 
| | KEY-UP   - QMK: KC_RCTL | Event key: Control    | Code: ControlRight | KeyCode: 17 |
| | KEY-UP   - QMK: KC_RSFT | Event key: Shift  |     Code: ShiftRight  |  KeyCode: 16 |
---

Whereas I'd expect the shift key to be unset before the ctrl+k key downs are sent.

## Attempted fixes

Below are resolutions I attempted with no success:

- Sending multiple `SS_DOWN(X_RSFT)` arguments
- Splitting the arguments into separate `SEND_STRING` commands
- Adding a delay between the `SEND_STRING` commands
- Using `clear_mods()`

coliss86, 2022-08-26:
What do you try to achieve ?
Have you tried to do the following : 
```
void TDShifter(qk_tap_dance_state_t *state, void *user_data) {
    if (state->count == 1) {
        SEND_STRING(SS_LCTL(SS_LSFT(SS_TAP(X_K))));
        return;
    }
}
```

But if you want to register the shift key, you can use : `register_mods(MOD_MASK_SHIFT);` and `unregister_mods(MOD_MASK_SHIFT);`

leep-frog, 2022-08-28:
The problem was that the "shift-up" code was getting sent at the end of the `SEND_STRING` command, even though the "shift-up" code was the first argument.  No clue why it was happening, but I updated my qmk fork and it's working as expected now.   

Closing.

closed 2022-08-28 by leep-frog