qmk/qmk_firmware #18130: [Bug] Container build script mismatching permissions?

Siilwyn, 2022-08-21:
Following the [getting started docker guide](https://github.com/qmk/qmk_firmware/blob/master/docs/getting_started_docker.md), running `util/docker_build.sh planck/rev6:default` with Podman on Fedora 36 results in the following error:
```sh
make: stat: Makefile: Permission denied
make: *** No rule to make target 'planck/rev6:default'.  Stop.
```

tzarc, 2022-11-01:
Can't reproduce it, I'm afraid:
```
% cat /etc/os-release
NAME="Fedora Linux"
VERSION="36 (Server Edition)"
ID=fedora
VERSION_ID=36
VERSION_CODENAME=""
PLATFORM_ID="platform:f36"
PRETTY_NAME="Fedora Linux 36 (Server Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:36"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f36/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=36
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=36
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Server Edition"
VARIANT_ID=server
```
```
% RUNTIME=podman ./util/docker_build.sh planck/rev6:default
QMK Firmware 0.18.16
Making planck/rev6 with keymap default

arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_planck_rev6/src/info_config.h                                                [OK]

<<<snip>>>

Linking: .build/planck_rev6_default.elf                                                             [OK]
Creating binary load file for flashing: .build/planck_rev6_default.bin                              [OK]
Creating load file for flashing: .build/planck_rev6_default.hex                                     [OK]

Size after:
   text    data     bss     dec     hex filename
      0   54876       0   54876    d65c planck_rev6_default.bin

Copying planck_rev6_default.bin to qmk_firmware folder                                              [OK]
(Firmware size check does not yet support STM32F303; skipping)
```
```
% stat planck_rev6_default.bin
  File: planck_rev6_default.bin
  Size: 54896           Blocks: 112        IO Block: 4096   regular file
```

tzarc, 2022-11-01:
Perhaps try:
```sh
qmk cd
find . -uid 0
```
If you get any files listed, then they're owned by `root` and you probably used `sudo` to run something previously.

Siilwyn, 2022-11-02:
@tzarc seems the difference is I'm using the workstation edition instead of the server:
```
NAME="Fedora Linux"
VERSION="36 (Workstation Edition)"
ID=fedora
VERSION_ID=36
VERSION_CODENAME=""
PLATFORM_ID="platform:f36"
PRETTY_NAME="Fedora Linux 36 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:36"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f36/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=36
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=36
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

`find . -uid 0` returns nothing.

Siilwyn, 2022-11-24:
The PR (https://github.com/qmk/qmk_firmware/pull/18932) got closed, how can we move this issue forward? I'm not sure what's wrong with adding the privileged flag in this scenario.

tzarc, 2022-11-24:
> The PR (#18932) got closed, how can we move this issue forward? I'm not sure what's wrong with adding the privileged flag in this scenario.

Can you show the output of: `./util/docker_cmd.sh stat Makefile` please?

Siilwyn, 2022-11-25:
@tzarc yes, thanks for the quick response!
```
  File: Makefile
  Size: 17291     	Blocks: 40         IO Block: 4096   regular file
Device: 28h/40d	Inode: 3636302     Links: 1
Access: (0644/-rw-r--r--)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-11-24 20:28:29.494610138 +0000
Modify: 2022-10-15 08:25:18.523531495 +0000
Change: 2022-10-15 08:25:18.523531495 +0000

```

tzarc, 2022-11-27:
Intriguing, looks like it's a `podman` thing:

```
$ RUNTIME=podman ./util/docker_cmd.sh stat Makefile
  File: Makefile
  Size: 17286           Blocks: 40         IO Block: 4096   regular file
Device: fc11h/64529d    Inode: 301108014   Links: 1
Access: (0644/-rw-r--r--)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-11-26 17:38:11.316327029 +0000
Modify: 2022-11-26 17:38:10.396338121 +0000
Change: 2022-11-26 17:38:10.396338121 +0000
 Birth: 2022-11-26 17:38:10.396338121 +0000

$ RUNTIME=docker ./util/docker_cmd.sh stat Makefile
  File: Makefile
  Size: 17286           Blocks: 40         IO Block: 4096   regular file
Device: fc11h/64529d    Inode: 301108014   Links: 1
Access: (0644/-rw-r--r--)  Uid: ( 1000/ UNKNOWN)   Gid: ( 1000/ UNKNOWN)
Access: 2022-11-26 17:38:11.316327029 +0000
Modify: 2022-11-26 17:38:10.396338121 +0000
Change: 2022-11-26 17:38:10.396338121 +0000
 Birth: 2022-11-26 17:38:10.396338121 +0000
```

Siilwyn, 2022-11-28:
That is odd indeed, thoughts on adding the privileged flag when podman is used?

tzarc, 2023-02-02:
> That is odd indeed, thoughts on adding the privileged flag when podman is used?

All build outputs would be owned by root, in that case. Not ideal once you're back outside the container.

Siilwyn, 2023-02-18:
@tzarc would've expected that too, you are talking about the bin files right? On inspection the bin files build with the privileged flags are still owned by me:
```
-rwxr-xr-x. 1 siilwyn siilwyn
```

So I guess adding the privileged with podman seems to be the way to go?

Eskers, 2023-07-14:
I'm also on Fedora, this looks to be an SELinux context issue rather than the container lacking any privileges.

Using the `z` option when initially mounting the volume fixes this without resorting to using the `--privileged` flag.

gabeklavans, 2023-09-11:
> I'm also on Fedora, this looks to be an SELinux context issue rather than the container lacking any privileges.
> 
> Using the `z` option when initially mounting the volume fixes this without resorting to using the `--privileged` flag.
> 
> Edit: A better overall solution maybe to update the documentation with instructions for the user to copy across the udev rules and remove the need for the use of `--privileged` at all?

Lifesaver, thank you. I was having the same issue. For a more concrete version of the solution, I changed my `docker run` command containing
```
-v host/path:container/path
```
to
```
-v host/path:container/path:z
```

tzarc, 2023-09-17:
If we can work out a command to test whether or not SELinux is in play and dynamically add it, sure.
Some systems won't have SELinux support so will likely bomb out.

freundTech, 2023-09-26:
> If we can work out a command to test whether or not SELinux is in play and dynamically add it, sure. Some systems won't have SELinux support so will likely bomb out.

I don't think that's needed. Both docker and podman just ignore the `:z` flag if SELinux is not used.

I just verified that the script still works with the flag on a clean Ubuntu 23.04 install in a VM (Ubuntu doesn't use SELinux), using both docker and podman from the ubuntu repos.
I also verified that adding the `:z` flag fixes the script on fedora silverblue 38.

arrowj, 2024-05-21:
I am running Fedora 40 and adding the :z flag to docker_build.sh is giving me no joy. Has there been any movement or adjustment to this issue in the last eight months? The error I recieve with or without the :z flag added is as follows:

```
QMK Firmware 0.24.11
make: *** No rule to make target 'keycrhon/q6/ansi_encoder:default'. Stop.
|
| QMK's make format is:
|     make keyboard_folder:keymap_folder[:target]
|
| Where `keyboard_folder` is the path to the keyboard relative to
| `qmk_firmware/keyboards/`, and `keymap_folder` is the name of the
| keymap folder under that board's `keymaps/` directory.
|
| Examples:
|     keyboards/dz60, keyboards/dz60/keymaps/default
|       -> make dz60:default
|       -> qmk compile -kb dz60 -km default
|     keyboards/planck/rev6, keyboards/planck/keymaps/default
|       -> make planck/rev6:default:flash
|       -> qmk flash -kb planck/rev6 -km default
|
```

The first time I ran it, it downloaded a bunch of blobs before throwing this error. Now it is just the error.

The command I am running is `util/docker_build.sh keycrhon/q6/ansi_encoder:default`

fauxpark, 2024-05-21:
You have a typo...

arrowj, 2024-05-21:
> You have a typo...

I was told that on Reddit and thought it was because I had not used copy/paste when writing the Reddit post. Turns out, I had made the same type twice. Thank you so much!

I removed the :z flag and it still compiled so I assume this was fixed somewhere and I am just an idiot.

PR #23988 (2024-06-24): Add support for userspace to docker build commands.
## Description

As per title.  Both `./util/docker_build.sh` and`./util/docker_cmd.sh` should be sorted, now. Simplified `_build` to call `_cmd` so there's no duplicate implementations.

Also adds two new hidden CLI commands -- `qmk path` and `qmk userspace-path` to print out the computed values for both `qmk_firmware` and `qmk_userspace`.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [x] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

##  [... cut]
closed 2024-06-24 by tzarc