qmk/qmk_firmware #18145: [Bug]  cannot find -lg_nano

sebastienb, 2022-08-23:
After a fresh install of QMK I can't generate a firmware for a planck v6. 

## Describe the Bug

When running `qmk compile -kb planck/rev6 -km default` I get this error right away

```
 qmk qmk compile -kb planck/rev6 -km default
Ψ Compiling keymap with gmake --jobs=1 planck/rev6:default

QMK Firmware 0.17.5
Making planck/rev6 with keymap default

arm-none-eabi-gcc (GCC) 12.1.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/command.c                                                                        [OK]
Linking: .build/planck_rev6_default.elf                                                             [ERRORS]
 |
 | /opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: cannot find -lg_nano
 | /opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: cannot find -lc_nano
 | collect2: error: ld returned 1 exit status
 |
gmake[1]: *** [builddefs/common_rules.mk:364: .build/planck_rev6_default.elf] Error 1
gmake: *** [Makefile:413: planck/rev6:default] Error 1
Make finished with errors
```

## System Information

**Keyboard:*planck * 
**Revision (if applicable):*rev 6* 
**Operating system:*Mac os Monterey* 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/sebs/qmk_firmware
Ψ Detected macOS 12.5.1.
Ψ Git branch: master
Ψ Repo version: 0.17.5
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.1.0
Ψ Found avr-gcc version 9.3.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

fauxpark, 2022-08-23:
You need `osx-cross/arm/arm-gcc-bin@8`, not whatever you installed.

closed 2022-08-29 by keyboard-magpie

maddocker, 2024-02-11:
I just had this issue, and I want to leave it here for reference for anyone else who stumbles across this. For more information, I am running Gentoo. My crossdev target for `arm-none-eabi` did not include the `newlib-nano` library, causing this error to be thrown when trying to compile a firmware. Gentoo users will need to enable the `nano` USE flag. This behavior appears to be included in the Gentoo-specific `qmk_firmware` install script, but my system privilege configuration was not allowing the install script to work as intended (probably my own fault; it's a learning process).