qmk/qmk_firmware #18153: GMMK Pro MacOS - can't REST

katiawheeler, 2022-08-24:
Apologies if this is not the right place. I flashed my GMMK Pro with the [MacOS keybindings](https://github.com/qmk/qmk_firmware/tree/master/keyboards/gmmk/pro/rev1/ansi/keymaps/macos) and am trying to use the RESET on the first layer but it's not working. Anyone have any suggestions on what I need to do?

keyboard-magpie, 2022-08-24:
If you are working with recent QMK you will need to change that keycode from `reset` to `QK_BOOT` for it to work.

katiawheeler, 2022-08-24:
> If you are working with recent QMK you will need to change that keycode from `reset` to `QK_BOOT` for it to work.

@keyboard-magpie I'm a QMK newbie - is there a way I can do that after having flashed it or will I need to find the manual reset button on the board?

keyboard-magpie, 2022-08-24:
Until you can edit the keymap and recompile the firmware, you will have to use the hardware reset button. I would suggest joining the qmk discord if you want help with the former. https://discord.gg/Z7pT7ycN