qmk/qmk_firmware #18166: [Bug] Keychron Q1 ANSI w/ encoder .hex cannot be flashed

davidmmiller, 2022-08-25:
## Describe the Bug

Compiling firmware for Keychron Q1 ANSI with encoder (rev0101) builds a .hex file.  QMK Toolbox says that only a .bin file can be flashed.

The rules.mk file for the keyboard rev includes:
    # MCU name
    MCU = atmega32u4
    
    # Bootloader selection
    BOOTLOADER = atmel-dfu
    // END

* QMK Toolbox 0.2.2 (https://qmk.fm/toolbox)
HID console connected: Keychron Keychron Q1 (3434:0107:0205)
STM32 DFU device connected: STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
Attempting to flash, please don't remove device
Only firmware files in .bin format can be flashed with dfu-util!
Flash complete 

## System Information

**Keyboard: Keychron Q1 ANSI v2 with encoder ** 
**Revision (if applicable): rev0101 in repo ** 
**Operating system: MacOS ** 
**`qmk doctor` output:** 

Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /Volumes/EXT-SSD/keyboards/qmk_firmware
Ψ Detected macOS 12.5.1.
Ψ Git branch: master
Ψ Repo version: 0.17.9
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-08-25 07:20:05 +1000 (fc0bf67f37) -- Add missing DD mapping for RGB_MATRIX_DRIVER (#18160)
Ψ - Latest upstream/master: 2022-08-25 07:20:05 +1000 (fc0bf67f37) -- Add missing DD mapping for RGB_MATRIX_DRIVER (#18160)
Ψ - Latest upstream/develop: 2022-08-25 12:17:41 +0100 (43fd647130) -- Add eeprom defaults for tinyuf2 bootloader (#18042)
Ψ - Common ancestor with upstream/master: 2022-08-25 07:20:05 +1000 (fc0bf67f37) -- Add missing DD mapping for RGB_MATRIX_DRIVER (#18160)
Ψ - Common ancestor with upstream/develop: 2022-08-25 07:20:05 +1000 (fc0bf67f37) -- Add missing DD mapping for RGB_MATRIX_DRIVER (#18160)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ QMK is ready to go, but minor problems were found

## Additional Context

% qmk compile -kb keychron/q1/rev_0101 -km davidmmiller
Ψ Compiling keymap with gmake --jobs=1 keychron/q1/rev_0101:davidmmiller

QMK Firmware 0.17.9
Making keychron/q1/rev_0101 with keymap davidmmiller

avr-gcc (Homebrew AVR GCC 8.4.0_2) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
[... 54 lines cut ...]
Compiling: platforms/avr/suspend.c                                                                  [OK]
Compiling: platforms/avr/timer.c                                                                    [OK]
Compiling: platforms/avr/bootloaders/dfu.c                                                          [OK]
Compiling: tmk_core/protocol/lufa/lufa.c                                                            [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Compiling: tmk_core/protocol/lufa/usb_util.c                                                        [OK]
Linking: .build/keychron_q1_rev_0101_davidmmiller.elf                                               [OK]
Creating load file for flashing: .build/keychron_q1_rev_0101_davidmmiller.hex                       [OK]
Copying keychron_q1_rev_0101_davidmmiller.hex to qmk_firmware folder                                [OK]
Checking file size of keychron_q1_rev_0101_davidmmiller.hex                                         [OK]
 * The firmware size is fine - 23114/28672 (80%, 5558 bytes free)

zvecr, 2022-08-26:
Not a bug, #16369 attempted to add support but Keychron seem more interested in using QMK as a selling tool than actually taking part in Open Source.

closed 2022-08-29 by keyboard-magpie

KiNG-GH, 2023-02-20:
Sorry to bother. But I still have this problem today while using qmk toolbox to flash latest firmware from via’s official. Have u figure it out??? Will the future firmware still be compatible to old Q1V2?

fonso003, 2023-03-02:
For anyone else following this, if you have a Q1 version 2 board all you have to do is download the firmware for the knob version of that board and flash it. The hardware is different. The firmware they have for adding your own knob is strictly for the version 1 board. 

https://cdn.shopify.com/s/files/1/0059/0630/1017/files/q1_us_knob_v2.6.bin?v=1675829311

PeterMortensen, 2024-06-09:
> The hardware is different.

Unless it is the "special edition" (SE), confusingly named Q1 V2 SE. It is a Q1 V1 in disguise ([AVR](https://www.microchip.com/wwwproducts/en/ATmega32U4)-based. For example, the official firmware for it is ***identical*** to the official firmware for Q1 V1 ANSI knob (only the file name is different).