qmk/qmk_firmware #18242: qmk setup fails on Fedora 36

onebitboy, 2022-09-01:
`qmk setup` fails on a fresh Fedora 36 installation with `usb-libusb.c:28:10: fatal error: usb.h: No such file or directory`.

Manually installing `libusb-devel` before running `qmk setup` fixes the error.

zvecr, 2022-09-01:
```
--2022-09-01 20:12:31--  https://www.obdev.at/downloads/vusb/bootloadHID.2012-12-08.tar.gz
Resolving www.obdev.at (www.obdev.at)... 78.46.114.187
Connecting to www.obdev.at (www.obdev.at)|78.46.114.187|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://www.obdev.at/ftp/pub/Products/vusb/bootloadHID.2012-12-08.tar.gz [following]
--2022-09-01 20:12:31--  https://www.obdev.at/ftp/pub/Products/vusb/bootloadHID.2012-12-08.tar.gz
Reusing existing connection to www.obdev.at:443.
HTTP request sent, awaiting response... 200 OK
Length: 107970 (105K) [application/x-gzip]
Saving to: ‘STDOUT’

-                                                           100%[========================================================================================================================================>] 105.44K  --.-KB/s    in 0.08s   

2022-09-01 20:12:31 (1.26 MB/s) - written to stdout [107970/107970]

gcc  -O2 -Wall `libusb-config --cflags` -c main.c -o main.o
/bin/sh: line 1: libusb-config: command not found
gcc  -O2 -Wall `libusb-config --cflags` -c usbcalls.c -o usbcalls.o
/bin/sh: line 1: libusb-config: command not found
In file included from usbcalls.c:19:
usb-libusb.c:28:10: fatal error: usb.h: No such file or directory
   28 | #include <usb.h>
      |          ^~~~~~~
compilation terminated.
make: *** [Makefile:46: usbcalls.o] Error 1
```

While a failure, isnt considered a hard failure within `qmk setup`. This would only be an issue if you need to use the `bootloadHID` tool for flashing.

fauxpark, 2022-09-02:
Would be nice if we could get packages added for bootloadhid on all the distros we support, so we don't have to compile it ourselves. At least there's only the executable to remove if you ever want to "uninstall" it.

PR #18243 (2022-09-02): Install correct package on Fedora

## Description

An alternative would be to patch bootloadHID source to link to `libusb-1.0.so` instead.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #18242

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [... cut]
closed 2022-09-02 by drashna