qmk/qmk_firmware #18245: Make quantum painter/pillow optional

dmlary, 2022-09-02:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
With the addition of quantum painter, and its dependency on pillow, the required packages for macos now requires the full stack of tcl-tk and X11 packages when installing via homebrew.  This is a lot of overhead for a command-line utility to build firmware and flash a keyboard.  Even more so when the keyboard doesn't make use of quantum painter.

Is there any way quantium painter, or specifically the pillow dependency, can become optional?  My hope is that if so the downstream homebrew recipe can offer an `--no-x11` install flag.

dmlary, 2022-12-01:
not stale; still looking for an answer on this.

tzarc, 2022-12-01:
Well, we'll welcome a PR. 

Doing what you're requesting is low priority compared to other items in the TODO list, and likely won't be handled by the collaborators any time soon.

Let us know if you're keen to take it up, otherwise we can just close it as "wontfix".

dmlary, 2022-12-01:
I’m open to creating the PR, but will need a little guidance for how deep the quantum painter integration goes, or just the pillow piece.  I opened this issue as soon as I noticed the dependency change in the hopes that whoever made the change would be able to provide some guidance of how to make it optional. 

Digging into the commit history, it looks like your experience with quantum painter probably makes you the best person to give that guidance.  I’d appreciate any suggestions you can provide about where & how pillow and quantum painter integrate, both in the code and the build system. Ideally it should be possible to ifdef out just the pillow pieces, but if not, would it be possible to do for quantum painter as a whole? I’m open to any suggestions you have. 

I’m happy to and capable of doing the grunt work, just need to know where to start looking.

tzarc, 2022-12-01:
Yes, I'd be best placed to consult -- I wrote the painter side of things.

Build system and C code should be unaffected.

All the pillow dependencies are local to the painter files within `lib/python`.
You'd have to work out how to safely lazy-load pillow and ensure the rest of the CLI works without it.

You may need to do some work to get things like the tests working with that scenario -- if pillow is no longer a dependency then the testing workflows will likely need it manually added... though I'd ignore that for now, until it becomes a problem in the GitHub action.

dmlary, 2023-03-06:
not stale, just need to get to it.

tzarc, 2023-03-06:
> not stale, just need to get to it.

Given there's no indication of _when_ that might be, I'm going to go ahead and close this issue. Feel free to raise a Pull Request when you get a chance to look at it.

closed 2023-03-06 by tzarc