qmk/qmk_firmware #18277: [Bug] Mouse Buttons Not Working When PS/2 is Enabled

galvy0, 2022-09-05:
## Describe the Bug
I'm testing the trackpoint functionality using handwired/trackpoint on the Elite-C v4.  
However, the mouse key functionalities are not working with the PS/2 enabled.  

## System Information

**Keyboard:** Elite-C microcontroller
**Revision (if applicable):** V4
**Operating system:** Windows 10
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/user/qmk_firmware
Ψ Detected Linux (WSL).
Ψ Git branch: galvy
Ψ Repo version: 0.16.9
Ψ - Latest galvy: 2022-09-05 13:02:41 -0400 (f15ddc9a92) -- try int
Ψ - Latest upstream/master: None
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: None
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.2.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
None

## Additional Context

Based on the /handwired/trackpoint keymap, the left mouse button is mapped to F0 and F1 pins.  
When I do `PS2_MOUSE_ENABLE = no` in rules.mk, connecting F0 and F1 pins triggers left click.  
When I do `PS2_MOUSE_ENABLE = yes`, connecting the pins does not trigger left click.  
To isolate the problem, I am trying it with just the bare Elite-C microcontroller (without the trackpoint module connected) and verified that the function still does not work (also tried with it connected with no luck).

klesh, 2022-09-25:
I'm using a blackpill micro controller, similar situations happened to me as well.
However, the mouse keys work sometimes if I moved the cursor into some places, it feels like sth is occupying the device, one has to move the cursor to some points for the firmware to release the device (wild guest).

drashna, 2022-09-27:
@klesh does this work better: 
https://github.com/drashna/qmk_firmware/commit/c253842bfb325507698df829cfe1d953064a98e2

klesh, 2022-09-27:
> @klesh does this work better: [drashna@c253842](https://github.com/drashna/qmk_firmware/commit/c253842bfb325507698df829cfe1d953064a98e2)

It seems like the commit is dangling, I was unable to fetch it from your repo.
I figured https://github.com/drashna/qmk_firmware/commit/b5fed749c34d5e9f18cb1f6687b666be2398c106 is the latest one. So I cherry-picked it and load it up to my keyboard.

It works, however, it shares the same problem with my fix: the middle button scrolling would send out the `middle button event` to the host, which in Linux, causes a paste action to be carried out 😹

galvy0, 2022-09-27:
@klesh I didn't find the root cause and a permanent fix (just not knowledgeable enough), so this is definitely still an issue.
But I did find a workaround that makes the mouse movement work at least, and it's quite simple.
https://github.com/galvy0/qmk_firmware/commit/29c43773a9cad8a2382341d598e5b3fc7dac355e

It still messes up when I try to use scroll button function with the trackpoint, so your problem with middle button scrolling is still there, but I figured for now I can just disable the scroll button function and live with it. I hope someone can fix this eventually...

drashna, 2022-09-27:
Sorry, the branch should be: 
https://github.com/drashna/qmk_firmware/tree/mouse_keys

Aside from the scroll issue, does this work better for you?

klesh, 2022-09-28:
> Sorry, the branch should be: https://github.com/drashna/qmk_firmware/tree/mouse_keys
> 
> Aside from the scroll issue, does this work better for you?

@drashna No problem, I figured it out yesterday, that was the branch I tested, the commit-shas are identical. It didn't work, the buttons were not working most of the time 😢

@galvy0 Yes, I agree, there are different kinds of devices, mousekey(from keyboard), ps/2 mouse(trackpoint), pointing device(trackball?)... makes it quite hard to figure out the problem, especially without the historical knowledge.

drashna, 2022-09-28:
Dang.  Does your change ontop of mine work?

klesh, 2022-09-28:
> Dang. Does your change ontop of mine work?

No, I commented them out.

drashna, 2022-09-29:
Taking a closer look, it looks like the ps2_mouse.c code could/should be updated to use `has_mouse_report_changed`, as this would help eliminate some of the issues.

In the meanwhile, it may work to change `has_mouse_report_changed` to check for the tp_buttons stuff. 

```c
extern int tp_buttons;
bool has_mouse_report_changed(report_mouse_t* new_report, report_mouse_t* old_report) {
	static int old_tp_buttons = 0;
    bool changed = (old_tp_buttons == tp_buttons);
    old_tp_buttons = tp_buttons;
    return memcmp(new_report, old_report, sizeof(report_mouse_t)) || changed;
}
#endif
```

Could throw this into the ps_mouse.c file, and see if that helps.

klesh, 2022-10-01:
> Taking a closer look, it looks like the ps2_mouse.c code could/should be updated to use `has_mouse_report_changed`, as this would help eliminate some of the issues.
>
> In the meanwhile, it may work to [...]

Good job, it works, except for the middle button scrolling sends out the middle click event.

drashna, 2022-10-04:
that's ... odd!

I'm not very familiar with the code, so I'm not sure why that would happen.  But it may be worth PRing this to the ps2 code, in general.

klesh, 2022-10-06:
Hi, all, I think I finally figured it out.
The `ps2_mouse_task` would return early if `pbuf_has_data` retuned a falsy condition causing `mosuekey` buttons to be ignored.

With the updated PR #18474 , my trackpoint is now working perfectly, the middle button scrolling, left click, and right click is all working nicely. 

Would someone try my fix to confirm?

galvy0, 2022-10-19:
@klesh sorry I've been swamped lately and just got around to trying out your PR. The TP works, but mouse scroll still doesn't work (PS2_MOUSE_ENABLE_SCROLLING). Did you get it to work with this enabled as well?

klesh, 2022-10-19:
@galvy0 Yes, my middle-button-scrolling is working perfectly.
However, I don't have `PS2_MOUSE_ENABLE_SCROLLING` setting whatsoever, it just works.
Here is my configuration for the `cantor` keyboard: https://github.com/klesh/qmk_firmware/tree/cantor/keyboards/cantor

drashna, 2022-12-10:
Should be fixed.

closed 2022-12-10 by drashna