qmk/qmk_firmware #18300: [Feature Request] Add option similar to STRICT_LAYER_RELEASE but for mouse keys.

itsdogtime, 2022-09-07:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

[Mouse buttons are cleared on layer change.](https://github.com/qmk/qmk_firmware/blob/master/quantum/action.c#L1078-L1081) However, holding down buttons can cause strain for people with carpal tunnel. This is difficult when applications require a mouse button be held often.

Could you please add an option similar to `#define STRICT_LAYER_RELEASE` but for mouse keys so that the mouse button won't be cleared on layer change?

drashna, 2022-10-04:
Ooooooh, that's why/where that happens!

drashna, 2022-12-04:
Closing. This should be fixed now.

PR #18577 (2022-11-12): Don't clear keys on layer change unless STRICT_LAYER_RELEASE is enabled
## Description

Previous behavior clears mousekeys on layer change, which is not expected/proper behavior. 

Not sure if extrakeys should be moved too?

## Types of Changes

- [x] Core
- [x] Bugfix
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #18300

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** d [... cut]
closed 2022-12-10 by drashna