qmk/qmk_firmware #18303: [Bug] qmk 0.18.x could not run on WeAct Black Pill V3.0 STM32F411CEU6

cavoirom, 2022-09-07:
## Describe the Bug

[My keyboard](https://github.com/cavoirom/qmk_firmware/tree/keyboard/mint/keyboards/ngxv/mint) is running QMK just fine on version _**0.17.9**_, I recently update to _**0.18.0**_, the firmware could build OK, but the keyboard is not working after flashing, the computer could not detect keyboard after connected. I tried 0.18.5 but no luck.

On my fork of QMK, I based on 0.17.9 and the firmware is working. You could try to build the keyboard with newest QMK to see it's not working.

I could don't know how to debug the keyboard

## System Information

**Keyboard:**  [ngxv/mint](https://github.com/cavoirom/qmk_firmware/tree/keyboard/mint/keyboards/ngxv/mint) 
**Revision (if applicable):** 
**Operating system:** I build the firmware on Linux within Docker. Keyboard connect to MacOS.

**Any keyboard related software installed?** No

zvecr, 2022-09-07:
`rules.mk` needs:

```makefile
BOARD = BLACKPILL_STM32_F411
```

cavoirom, 2022-09-10:
Thanks @zvecr, I fixed it with you suggestion. Do you know when to find the document about `BOARD`'s value? or I need to take the example from other keyboards?

closed 2022-09-20 by cavoirom