qmk/qmk_firmware #18325: Fatal error while compiling

csolje, 2022-09-10:
Hi,
I am on a cantor 42 keys keeb and I am trying to compile my firmware and I get a fatal error: printf.h: No Such file or directory
As upi can see on this screanshoot.
[image]

I am running version 0.18.5 of qmk, and I don't know what I am doing wrong.

drashna, 2022-09-10:
Run "make git-submodule" from the qmk_firmware folder. 

The print code was moved to a submodule, and you may need to update them.

csolje, 2022-09-10:
I have done that already and it still gives me that error

csolje, 2022-09-10:
[image]

csolje, 2022-09-10:
Yeah I had it on both and now I found the issue and you where right the submodules need updating and I also had an error in my rules.c file so I will be closing this issue it is working on both now

closed 2022-09-10 by csolje