qmk/qmk_firmware #18331: [Feature Request] Limiting combos to certain layers

xsrvmy, 2022-09-11:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ X ] Alteration (enhancement/optimization) of existing feature(s)
- [ X ] New behavior

## Description
Currently there is no simple way disable combos on layers, if the layer has the same keys that have combos on the base layer, or if layer independent combos are enabled. C code has to be used instead.

It would be nice to be able to specify in the code that combos should only be used on some layers. Similar to other settings, there could be one setting for all combos, and one setting for only specific combos. The main use case for this is to disable combos on non-alpha layers and gaming layer without resorting to C code.

icecreammatt, 2022-12-26:
I'd love to have this feature too. I currently use `d+k` as the `return` key for `qwerty` mode on my keyboard but use `s+e` in `colemak` mode while transitioning and when I go back to `qwerty` mode I keep pressing return when typing words that end in `es` by accident.

xsrvmy, 2023-01-13:
Hmm I have noticed that `combo_should_trigger` exists already. I guess this is probably the easiest way to implement this for now.

closed 2023-05-14 by github-actions[bot]

MrScottyTay, 2024-01-26:
Is nothing going to come of this? Really sucks to go onto a "gaming" layer, but having your typing/coding combos to get in the way of playing games

MrScottyTay, 2024-01-26:
ah, thank you