qmk/qmk_firmware #18336: Having trouble compiling for Keychron Q5 on mac... M1 Mini

BLTHSHZZR, 2022-09-11:
### Information

- [X] None of the other issue templates apply

### Description

This isn't really a bug...I don't think, so I'm submitting it as "Other"

I downloaded the repository with the firmware for my keyboard (Keychron Q5) here (actually the playground repository branch since it has my actual model): https://github.com/Keychron/qmk_firmware.git 

When I attempt to compile by using 

> qmk compile -kb keychron/q5/q5_ansi_stm32l432_ec11 -km via

Or even when I copy-paste the firmware folder for my Q5 model itself, this is the message that returns:

```
Compiling keymap with gmake --jobs=1 /Users/user/Desktop/qmk_firmware-playground/keyboards/keychron/q5/q5_ansi_stm32l432_ec11:via

QMK Firmware 0.18.6
make: *** No rule to make target '/Users/user/Desktop/qmk_firmware-playground/keyboards/keychron/q5/q5_ansi_stm32l432_ec11:via'. Stop.
|
| QMK's make format is:
|     make keyboard_folder:keymap_folder[:target]
|
| Where `keyboard_folder` is the path to the keyboard relative to
| `qmk_firmware/keyboards/`, and `keymap_folder` is the name of the
| keymap folder under that board's `keymaps/` directory.
|
| Examples:
|     keyboards/dz60, keyboards/dz60/keymaps/default
|       -> make dz60:default
|       -> qmk compile -kb dz60 -km default
|     keyboards/planck/rev6, keyboards/planck/keymaps/default
|       -> make planck/rev6:default:flash
|       -> qmk flash -kb planck/rev6 -km default
|
user@user-Mac-mini ~ % make keychron/q5/q5_ansi_stm32l432_ec11:via
make: *** No rule to make target `keychron/q5/q5_ansi_stm32l432_ec11:via'.  Stop.
user@user-Mac-mini ~ % make /Users/user/Desktop/qmk_firmware-playground/keyboards/keychron/q5/q5_ansi_stm32l432_ec11 :via
make: Nothing to be done for `/Users/user/Desktop/qmk_firmware-playground/keyboards/keychron/q5/q5_ansi_stm32l432_ec11'.
make: *** No rule to make target `:via'.  Stop.
user@user-Mac-mini ~ % qmk compile -kb keychron/q5/q5_ansi_stm32l432_ec11 -km via

usage: qmk compile [-h] [-c] [-e ENV] [-j PARALLEL] [-n] [-km KEYMAP] [-kb KEYBOARD]
                   [filename]
qmk compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'keychron/q5/q5_ansi_stm32l432_ec11'
user@user-Mac-mini ~ % 
```

I'm relatively new to UNIX and QMK so I'm not even entirely clear what it's telling me the problem is. Help?

fauxpark, 2022-09-11:
When you run `make`, you need to be in the `qmk_firmware` directory. It's not a QMK-specific tool; it simply looks for a `Makefile` in the current directory. The QMK CLI is a little smarter, but it will default to looking in the `qmk_firmware` directory set as your QMK home when you ran `qmk setup`. If you want to use another fork, you'll need to `cd` into it first.

As the error message shows, the keyboard name is the folder name relative to the `qmk_firmware/keyboards` directory. You don't give it a full path.

BLTHSHZZR, 2022-09-11:
> 

I'm very grateful someone knows what I'm doing wrong :)

I'm really unsure what I set as my QMK home when I ran the initial setup...I didn't actually manually set anything and I can't find any folder anywhere in my library or otherwise called QMK

Okay...so in much simpler terms...what do I need to do with the repo (it's just a folder sitting on my desktop), or what do I need to type into my terminal to give it a full path?

fauxpark, 2022-09-11:
`qmk doctor` will tell you where your QMK home is. By default the repo gets cloned into your home directory - `~/qmk_firmware` on macOS and Linux, `%USERPROFILE%/qmk_firmware` for Windows.

> Okay...so in much simpler terms...what do I need to do with the repo (it's just a folder sitting on my desktop), or what do I need to type into my terminal to give it a full path?

If you downloaded the .zip from GitHub, that's not going to work. It needs to be cloned with Git. Run `qmk clone -h`.

BLTHSHZZR, 2022-09-12:
> `qmk doctor` will tell you where your QMK home is. By default the repo gets cloned into your home directory - `~/qmk_firmware` on macOS and Linux, `%USERPROFILE%/qmk_firmware` for Windows.
>
>
> If you [...]

Ok. I think I've got my environment set. I only have one qmk_firmware file folder on my computer and I can locate it in my library. My Keyboard is a Keychron Q5 with knob, so the correct folder with the files in it that seemingly comprise the firmware is: Users/user/qmk_firmware/keyboards/keychron/q5/q5_ansi_stm32l432_ec11

qmk is installed and runs in terminal.
I am using the command:

` qmk compile -kb /Users/user/qmk_firmware/keyboards/keychron/q5/q5_ansi_stm32l432_ec11  -km via'

with this as the result:

Ψ Compiling keymap with gmake --jobs=1 /Users/user/qmk_firmware/keyboards/keychron/q5/q5_ansi_stm32l432_ec11:via

QMK Firmware 0.18.6
make: *** No rule to make target '/Users/user/qmk_firmware/keyboards/keychron/q5/q5_ansi_stm32l432_ec11:via'. Stop.
|
| QMK's make format is:
|     make keyboard_folder:keymap_folder[:target]
|
| Where `keyboard_folder` is the path to the keyboard relative to
| `qmk_firmware/keyboards/`, and `keymap_folder` is the name of the
| keymap folder under that board's `keymaps/` directory.
|
| Examples:
|     keyboards/dz60, keyboards/dz60/keymaps/default
|       -> make dz60:default
|       -> qmk compile -kb dz60 -km default
|     keyboards/planck/rev6, keyboards/planck/keymaps/default
|       -> make planck/rev6:default:flash
|       -> qmk flash -kb planck/rev6 -km default
|
user@user-Mac-mini ~ % 
`

As though I am a monkey who has learned to follow English language orders very literally, what should I do at this point to compile the firmware properly? (my ultimate goal is to unlock some additional rgb settings that already appear in the "config.h" of the parent folder (q5) that also has a folder with my model inside of it (the folder called q5_ansi_stm32l432_ec11).

I understand that on some level I am using the wrong command structure but I have no idea where to go from here. I have attempted to attach an image of the file directory I'm working with if that helps

[image]

closed 2022-09-18 by BLTHSHZZR