qmk/qmk_firmware #18357: Can the trigger resistance between col and row be reduced? For example, now it is less than 10KΩ and nine triggers, and I want it to trigger when it is less than 5KΩ. Can I modify the code to implement it or can I only change the hardware? I want to use the joystick as an ordinary four-way switch. Now it will always be in the trigger state. Another open source project can be used directly and cannot be always triggered.

jeromeli002, 2022-09-14:
Can the trigger resistance between col and row be reduced? For example, now it is less than 10KΩ and nine triggers, and I want it to trigger when it is less than 5KΩ. Can I modify the code to implement it or can I only change the hardware? I want to use the joystick as an ordinary four-way switch. Now it will always be in the trigger state. Another open source project can be used directly and cannot be always triggered.

daskygit, 2022-09-14:
No, you can't change the internal pull-up so you're limited to working with the equivalent resistance this provides. You're likely best make using of an ADC (https://docs.qmk.fm/#/adc_driver?id=adc-driver) and custom matrix. This would allow to change how far you need to move the joystick to trigger a key press.

precondition, 2022-09-14:
That's quite a long title

closed 2023-01-06 by zvecr