qmk/qmk_firmware #18387: [Bug] compile error usb_descriptor.c:448:35: error: 'DEVICE_VER' undeclared

osamuaoki, 2022-09-17:
I experience compile error if I use the latest develop branch to compile for my handwired keyboard cgs50.

* 786fabc144 ("Merge remote-tracking branch 'origin/master' into develop", 2022-09-16) -- error as below
* bc0756f294 ("Disconnect `usb.device_ver` (#18259)", 2022-09-16) -- error   
   * bad commit?
   * oldest commit I found error
   *  this is python code
* 23e12497b2 ("Merge remote-tracking branch 'origin/master' into develop", 2022-09-12) -- no problem

```
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_cgs50/src/info_config.h                                                      [OK]
...
Compiling: tmk_core/protocol/chibios/chibios.c                                                      [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                      tmk_core/protocol/usb_descriptor.c:448:35: error: 'DEVICE_VER' undeclared here (not in a function)
  448 |     .ReleaseNumber              = DEVICE_VER,
      |                                   ^~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:359: .build/obj_cgs50_pico/usb_descriptor.o] Error 1
Make finished with errors
make: *** [Makefile:414: cgs50:pico] Error 1
```

I set info.json as:

```
{
    "keyboard_name": "cgs50",
    "manufacturer":"osamuaoki",
    "url": "https://github.com/osamuaoki/qmk_firmware/tree/osamu1/keyboards/cgs50",
    "maintainer": "Osamu Aoki <osamu@debian.org>",
    "usb": {
        "vid": "0xFEED",
        "pid": "0x0000",
        "device_ver": "0x0002"
    },
}                                                                                    
```

**Any keyboard related software installed?** 
 - [x] None

zvecr, 2022-09-17:
Since https://github.com/qmk/qmk_firmware/pull/18259, `device_ver` is no longer valid,
```diff
-        "device_ver": "0x0002"
+        "device_version": "0.0.2"
```

osamuaoki, 2022-09-17:
I see but can this python code parse string to emit reasonable message?
Unlike changes required to old fashoned Makefile, this should be easier.

As for MANUFACTURER etc. in Makefile, ignoring them and error out with clear message to migrate to info.json helps.

osamuaoki, 2022-09-17:
Hmmm... I thought I tested new version style too.
It errors with
```
{
    "keyboard_name": "cgs58",
    "manufacturer":"osamuaoki",
    "url": "https://github.com/osamuaoki/qmk_firmware/tree/osamu1/keyboards/cgs58",
    "maintainer": "Osamu Aoki <osamu@debian.org>",
    "usb": {
        "vid": "0xFEED",
        "pid": "0x0000",
        "device_ver": "0.0.2"
    },
}                                                     
```

osamuaoki, 2022-09-17:
Oops, wrong file... I see my problem.  Sorry.

osamuaoki, 2022-09-17:
With correct file: I still get errors
```
☒ Not including data from file: keyboards/cgs50/info.json
☒ 	usb.device_ver: '0.0.2' does not match '^0x[0-9A-F]{4}$'
Making cgs50 with keymap pico and target flash

☒ Not including data from file: keyboards/cgs50/info.json
☒ 	usb.device_ver: '0.0.2' does not match '^0x[0-9A-F]{4}$'
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_cgs50/src/info_config.h                                                      [WARNINGS]
 | 
 | ☒ Not including data from file: keyboards/cgs50/info.json
 | ☒ 	usb.device_ver: '0.0.2' does not match '^0x[0-9A-F]{4}$'
 | 
Generating: .build/obj_cgs50/src/default_keyboard.c                                                 [WARNINGS]
 | 
 | ☒ Not including data from file: keyboards/cgs50/info.json
 | ☒ 	usb.device_ver: '0.0.2' does not match '^0x[0-9A-F]{4}$'
 | 
Generating: .build/obj_cgs50/src/default_keyboard.h                                                 [WARNINGS]
 | 
 | ☒ Not including data from file: keyboards/cgs50/info.json
 | ☒ 	usb.device_ver: '0.0.2' does not match '^0x[0-9A-F]{4}$'
 | 
Generating: .build/obj_cgs50/src/layouts.h                                                          [WARNINGS]
 | 
 | ☒ Not including data from file: keyboards/cgs50/info.json
 | ☒ 	usb.device_ver: '0.0.2' does not match '^0x[0-9A-F]{4}$'

```

osamuaoki, 2022-09-17:
It looks like old sanity check code preventing to use new version style.

zvecr, 2022-09-17:
`device_version` not `device_ver`

osamuaoki, 2022-09-17:
Thanks. I see.  Well, I guess this is develop branch.

closed 2022-09-17 by zvecr