qmk/qmk_firmware #18390: [Bug] firmware not working in 0.18.x but ok with 0.17.9

alinelena, 2022-09-17:
I have this converted tipros, that use a stm32f401 and a shift register to play with some leds.
qmk code is in here.
https://github.com/alinelena/qmk_firmware/tree/mlego/keyboards/tipro

firmware generated with 
```
qmk compile -kb tipro/rev2 -km default_96 
```
once copied on blackpill works fine with 0.17.9, but not if I use current devel...or any of 0.18.x
practically I can flash it but does not show up on usb

daskygit, 2022-09-17:
The default board configuration for stm32f4x1 has been changed with the last develop merge. You can see your previous boards have been updated with the new config option in the repo.

[`BOARD = BLACKPILL_STM32_F401`](https://github.com/qmk/qmk_firmware/blob/4781a798cab353397cf2ab92908a49096d505891/keyboards/mlego/m48/rev1/rules.mk#L3)

alinelena, 2022-09-17:
> The default board configuration for stm32f4x1 has been changed with the last develop merge. You can see your previous boards have been updated with the new config option in the repo.
>
> [`BOARD = [...]

thank you that was.
board was missing in my info.json... no idea how this happened.

closed 2022-09-17 by zvecr