qmk/qmk_firmware #18395: Need Assistance Compiling Firmware/Updating config.h for RGB lighting effects for Keychron Q5 from repository on Mac OS Monterey

BLTHSHZZR, 2022-09-18:
Okay. I have a Keychron q5. Very nice build.
I have the firmware folder downloaded for it. originally i downloaded it straight from the page https://github.com/Keychron/qmk_firmware/tree/playground
but later I used the command "git clone -b playground https://github.com/Keychron/qmk_firmware.git" to make it show up as my standard qmk_firmware folder in my library.

I have some issues, which hopefully the attached image can help explain.

1. There are four models of keychron represented by the lowest set of folders. I have the ANSI knob version (shown here as "q5_ansi_stm32l432_ec11"). Should I just delete those others or will that cause problems?

2. The whole reason I'm trying to learn qmk is so that I can change the rainbow vomit lighting settings. I already worked through the whole "delete the // before "define" in the config.h file for the main q5 folder (boxed in green). There are two other files that end in ".h" (boxed in red and blue). Neither of these have the lighting settings in them and instead just seem to show a big list of the keys/rgbs. Do I need to do anything with them?

3. Finally, what am I "compiling" and how? I imagine I would just try to compile the folder for my keyboard...but if the "config.h" file with the actual rgb settings in it is in the main "q5" folder...won't compiling have zero effect on my lighting stuff? I just need some help here to figure out what the heck is going on and what I need to do at this point.

Running on a Mac, Monterey, 12.5
[image]

adophoxia, 2022-09-18:
1. You can just leave them there as they are. 
2. (without going technical) the "[keyboard_name].c" file stores info about the locations of the LEDs on the PCB while "[keyboard_name].h" refers to the matrix locations of the switches on the PCB. For the lighting effects you're looking for, it's in the first "config.h" file you boxed in green, which you have to scroll down the bottom to look for.
3. You're compiling the firmware that you are building to flash onto the keyboard itself. Once you've made the changes made for your keyboard, you'd have to compile using "make keychron/q5/q5_stm32l432_ec11:[name_of_keymap]". After that, you'd have to put the board into bootloader mode, which for Keychron boards, you'd have to press down the reset button that's under the spacebar while plugging in the USB-C cable. You then use QMK Toolbox to load and flash the ".bin" file for the firmware you just compiled.

BLTHSHZZR, 2022-09-18:
> 

So I Imagine where "Keymap Name" appears I just designate my own name so I can find it. Where will this firmware show up? Same folder?

BLTHSHZZR, 2022-09-18:
All Right so these are the results I'm getting. Where am I screwing up?

`Last login: Sat Sep 17 19:09:14 on ttys000
USER@USER-Mac-mini ~ % qmk compile -kb keychron/q5/q5_stm32l432_ec11 -km keymapLITES
usage: qmk compile [-h] [-c] [-e ENV] [-j PARALLEL] [-n] [-km KEYMAP]
                   [-kb KEYBOARD]
                   [filename]
qmk compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'keychron/q5/q5_stm32l432_ec11'
USER@USER-Mac-mini ~ % make keychron/q5/q5_stm32l432_ec11:keymapLITE
make: *** No rule to make target `keychron/q5/q5_stm32l432_ec11:keymapLITE'.  Stop.
USER@USERs-Mac-mini ~ % 
`

adophoxia, 2022-09-18:
Are you sure the name of the keymap is the correct one when you try and compile?

BLTHSHZZR, 2022-09-18:
> Are you sure the name of the keymap is the correct one when you try and compile?

I am not. I thought I needed to make up my own to distinguish it from the base. I suppose the name should be "via" and I am going to try that. 

[image]

closed 2023-01-17 by github-actions[bot]