qmk/qmk_firmware #18401: [Bug] Boardsource: Lulu: RP2040: RGB_ALPHAS_MODS appears to be configured wrong

jumper149, 2022-09-18:
Please give me some advice what I need to fix and I'll probably do the dirty work for you.

[image]

## Describe the Bug

I think only the bottom row should be set as mods.

## System Information

**Keyboard:** Boardsource Lulu
**Revision (if applicable):** https://github.com/jumper149/qmk_firmware/commit/9e904d356a1328399e7c5f2d778477adc02219a9
**Operating system:** 
**`qmk doctor` output:** 
```
(Paste output here)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

jumper149, 2022-09-18:
I made some progress using this diff:

```
diff --git a/keyboards/boardsource/lulu/info.json b/keyboards/boardsource/lulu/info.json
index 3418cd6e7a..e2ef079a6d 100644
--- a/keyboards/boardsource/lulu/info.json
+++ b/keyboards/boardsource/lulu/info.json
@@ -36,66 +36,66 @@
       { "flags": 2, "x": 17, "y": 10 },
       { "flags": 2, "x": 51, "y": 10 },
       { "flags": 2, "x": 86, "y": 10 },
+      { "flags": 4, "matrix": [0, 5], "x": 86,  "y": 0 },
+      { "flags": 4, "matrix": [0, 4], "x": 68,  "y": 0 },
+      { "flags": 4, "matrix": [0, 3], "x": 51,  "y": 0 },
+      { "flags": 4, "matrix": [0, 2], "x": 34,  "y": 0 },
+      { "flags": 4, "matrix": [0, 1], "x": 17,  "y": 0 },
+      { "flags": 1, "matrix": [0, 0], "x": 0,  "y": 0 },
+      { "flags": 1, "matrix": [1, 0], "x": 0,  "y": 16 },
+      { "flags": 4, "matrix": [1, 1], "x": 17,  "y": 16 },
+      { "flags": 4, "matrix": [1, 2], "x": 34,  "y": 16 },
+      { "flags": 4, "matrix": [1, 3], "x": 51,  "y": 16 },
+      { "flags": 4, "matrix": [1, 4], "x": 68,  "y": 16 },
+      { "flags": 4, "matrix": [1, 5], "x": 86,  "y": 16 },
+      { "flags": 4, "matrix": [2, 5], "x": 86,  "y": 32 },
+      { "flags": 4, "matrix": [2, 4], "x": 68,  "y": 32 },
+      { "flags": 4, "matrix": [2, 3], "x": 51,  "y": 32 },
+      { "flags": 4, "matrix": [2, 2], "x": 34,  "y": 32 },
+      { "flags": 4, "matrix": [2, 1], "x": 17,  "y": 32 },
+      { "flags": 1, "matrix": [2, 0], "x": 0,  "y": 32 },
+      { "flags": 1, "matrix": [3, 0], "x": 0,  "y": 48 },
+      { "flags": 4, "matrix": [3, 1], "x": 17,  "y": 48 },
+      { "flags": 4, "matrix": [3, 2], "x": 34,  "y": 48 },
+      { "flags": 4, "matrix": [3, 3], "x": 51,  "y": 48 },
+      { "flags": 4, "matrix": [3, 4], "x": 68,  "y": 48 },
+      { "flags": 4, "matrix": [3, 5], "x": 86,  "y": 48 },
+      { "flags": 4, "matrix": [4, 5], "x": 103,  "y": 48 },
+      { "flags": 1, "matrix": [4, 4], "x": 86,  "y": 64 },
+      { "flags": 1, "matrix": [4, 3], "x": 68,  "y": 64 },
+      { "flags": 1, "matrix": [4, 2], "x": 51,  "y": 64 },
+      { "flags": 1, "matrix": [4, 1], "x": 34,  "y": 64 },
       { "flags": 2, "x": 137, "y": 55 },
       { "flags": 2, "x": 172, "y": 55 },
       { "flags": 2, "x": 206, "y": 40 },
       { "flags": 2, "x": 206, "y": 10 },
       { "flags": 2, "x": 172, "y": 10 },
       { "flags": 2, "x": 137, "y": 10 },
-      { "flags": 1, "matrix": [0, 0], "x": 0,  "y": 0 },
-      { "flags": 4, "matrix": [0, 1], "x": 17,  "y": 0 },
-      { "flags": 4, "matrix": [0, 2], "x": 34,  "y": 0 },
-      { "flags": 4, "matrix": [0, 3], "x": 51,  "y": 0 },
-      { "flags": 4, "matrix": [0, 4], "x": 68,  "y": 0 },
-      { "flags": 4, "matrix": [0, 5], "x": 86,  "y": 0 },
       { "flags": 4, "matrix": [5, 5], "x": 137,  "y": 0 },
       { "flags": 4, "matrix": [5, 4], "x": 155,  "y": 0 },
       { "flags": 4, "matrix": [5, 3], "x": 172,  "y": 0 },
       { "flags": 4, "matrix": [5, 2], "x": 189,  "y": 0 },
       { "flags": 4, "matrix": [5, 1], "x": 206,  "y": 0 },
       { "flags": 1, "matrix": [5, 0], "x": 224,  "y": 0 },
-      { "flags": 1, "matrix": [1, 0], "x": 0,  "y": 16 },
-      { "flags": 4, "matrix": [1, 1], "x": 17,  "y": 16 },
-      { "flags": 4, "matrix": [1, 2], "x": 34,  "y": 16 },
-      { "flags": 4, "matrix": [1, 3], "x": 51,  "y": 16 },
-      { "flags": 4, "matrix": [1, 4], "x": 68,  "y": 16 },
-      { "flags": 4, "matrix": [1, 5], "x": 86,  "y": 16 },
-      { "flags": 4, "matrix": [6, 5], "x": 137,  "y": 16 },
-      { "flags": 4, "matrix": [6, 4], "x": 155,  "y": 16 },
-      { "flags": 4, "matrix": [6, 3], "x": 172,  "y": 16 },
-      { "flags": 4, "matrix": [6, 2], "x": 189,  "y": 16 },
-      { "flags": 4, "matrix": [6, 1], "x": 206,  "y": 16 },
       { "flags": 1, "matrix": [6, 4], "x": 224,  "y": 16 },
[... 20 lines cut ...]
-      { "flags": 4, "matrix": [3, 3], "x": 51,  "y": 48 },
-      { "flags": 4, "matrix": [3, 4], "x": 68,  "y": 48 },
-      { "flags": 4, "matrix": [3, 5], "x": 86,  "y": 48 },
-      { "flags": 4, "matrix": [4, 5], "x": 103,  "y": 48 },
-      { "flags": 4, "matrix": [9, 5], "x": 120,  "y": 48 },
-      { "flags": 4, "matrix": [8, 5], "x": 137,  "y": 48 },
-      { "flags": 4, "matrix": [8, 4], "x": 155,  "y": 48 },
-      { "flags": 4, "matrix": [8, 3], "x": 172,  "y": 48 },
-      { "flags": 4, "matrix": [8, 2], "x": 189,  "y": 48 },
-      { "flags": 4, "matrix": [8, 1], "x": 206,  "y": 48 },
       { "flags": 1, "matrix": [8, 4], "x": 224,  "y": 48 },
-      { "flags": 1, "matrix": [4, 1], "x": 34,  "y": 64 },
-      { "flags": 1, "matrix": [4, 2], "x": 51,  "y": 64 },
-      { "flags": 1, "matrix": [4, 3], "x": 68,  "y": 64 },
-      { "flags": 1, "matrix": [4, 4], "x": 86,  "y": 64 },
+      { "flags": 4, "matrix": [8, 1], "x": 206,  "y": 48 },
+      { "flags": 4, "matrix": [8, 2], "x": 189,  "y": 48 },
+      { "flags": 4, "matrix": [8, 3], "x": 172,  "y": 48 },
+      { "flags": 4, "matrix": [8, 4], "x": 155,  "y": 48 },
+      { "flags": 4, "matrix": [8, 5], "x": 137,  "y": 48 },
+      { "flags": 4, "matrix": [9, 5], "x": 120,  "y": 48 },
       { "flags": 1, "matrix": [9, 4], "x": 137,  "y": 64 },
       { "flags": 1, "matrix": [9, 3], "x": 155,  "y": 64 },
       { "flags": 1, "matrix": [9, 2], "x": 172,  "y": 64 },
```

jumper149, 2022-09-18:
I fixed it here: https://github.com/jumper149/qmk_firmware/tree/862a6118cae356b98bb4bcd03732e646e432397d

Once I'm done with my keymap etc. I'll upstream it.

[image]
[image]

daskygit, 2022-09-18:
The fix for the info.json should be submitted seperately from the keymap addition please.

PR #18407 (2022-09-19): Fix boardsource/lulu RGB matrix

I stumbled upon issue #18401.

Now the RGB matrix is in the correct order.
I fixed some other small details in separate commits.

The linked issue shows some pictures of the working firmware.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #18401

## Checklist

- [ ] (doesn't apply) My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/codi [... cut]
closed 2022-09-19 by zvecr