qmk/qmk_firmware #18435: [Bug] AnnePro2 - CapsLock Indicator not working in Bluetooth mode

bojanmilevski, 2022-09-20:
**Keyboard**: Anne Pro 2
**Version**: C18

When connected via USB, the CapsLock key indicates whether or not the key is activated, but when switching over to Bluetooth, the indicator completely stops functioning (the CapsLock key does not light up).

I have tested `annepro2/keymaps/default`, `annepro2/keymaps/default-full-caps` and `annepro2/keymaps/default-layer-indicators`, I have even tried [Open Anne Pro 2's QMK fork](https://github.com/OpenAnnePro/qmk_firmware/tree/keyboard-annepro2/keyboards/annepro2), but the issue is present throughout all these configurations.

I have scoured the internet for possible solutions, but I have not been successful. Many people have faced this same issue, yet none of them have an appropriate answer.

I would appreciate any help, thank you in advance!

fauxpark, 2022-09-20:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/annepro2/annepro2_ble.c#L128-L130

It seems like it's not implemented. You'll likely need to investigate how it works on the factory firmware.

bojanmilevski, 2022-09-20:
Just for clarification purposes, by "factory firmware", do you mean the proprietary firmware (ObinsKit firmware)? Thank you in advance.

mkopec, 2025-03-08:
See https://github.com/qmk/qmk_firmware/pull/24996