qmk/qmk_firmware #18457: [Bug] Chimera Ergo some keys don't work with default keymap

andrewgribben, 2022-09-23:
I decided to update the firmware on my Chimera Ergo from @jawn-smith fork in order to get the latest features and let me use the configurator but I found that with the new firmware several keys on the bottom row would not work.

## Describe the Bug

Using the latest build of QMK created via `util/docker_build.sh glenpickle/chimera_ergo:default` the Z, V and N keys would not work on the default keymap. Changing the layout in chimera_ergo.h to mirror what was used by the board's creator https://github.com/jawn-smith/qmk_firmware-1 seems to fix the issue.

```
#define LAYOUT( \
  k00, k01, k02, k03, k04, k05,      k06, k07, k08, k09, k10, k11, \
  k12, k13, k14, k15, k16, k17,      k18, k19, k20, k21, k22, k23, \
  k24, k25, k26, k27, k28, k29,      k30, k31, k32, k33, k34, k35, \
  k36, k37, k38, k39, k40, k41,      k42, k43, k44, k45, k46, k47, \
                      k48, k49,      k50, k51 \
) { \
{ k03, k40, k26, k15, k28, k01,      k42, k31, k20, k33, k08, k10     }, \
{ k00, k37, k14, k27, k16, k36,      k47, k19, k32, k21, k46, k11  }, \
{ k12, k25, k02, k39, k17, k49,      k50, k18, k44, k09, k34, k23  }, \
{ k24, k13, k38, k04, k05, k48,      k51, k06, k07, k45, k22, k35  },  \
{ k29, k41, k03, k40, k37, KC_G,         k30, k43, k08, k10, k46, KC_H     }, \
}

```

I've since found an open PR from @rclasen on the QMK fork, https://github.com/jawn-smith/qmk_firmware-1/pull/1/commits/2cde23e95600e8bd112df47e27902ec47bb3b3e9

I assume when the chimera ergo code was merged into QMK it was using the broken LAYOUT?

drashna, 2022-12-10:
If you have the actual board, then opening a PR to fix it would be appreciated.

plgrazon, 2023-11-11:
@andrewgribben did you get this to work? I'm having the same issue