qmk/qmk_firmware #18529: [Bug] wear_leveling_rp2040_flash_config.h path of POST_CONFIG_H looks wrong

hajimen, 2022-09-29:
branch: develop
file: `builddefs/common_features.mk`
commit: https://github.com/qmk/qmk_firmware/commit/7c1797f52f74c9614615c0632ea1a2f5f11d3af6
line: 275

```
POST_CONFIG_H += $(DRIVER_PATH)/wear_leveling/wear_leveling_rp2040_flash_config.h
```

This line makes `drivers/wear_leveling/wear_leveling_rp2040_flash_config.h` path, but the file is not exist there. Isn't it `platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash_config.h`? If so, the line should be:

```
POST_CONFIG_H += $(PLATFORM_PATH)/$(PLATFORM_KEY)/$(DRIVER_DIR)/wear_leveling/wear_leveling_rp2040_flash_config.h
```

zvecr, 2022-09-29:
`$(PLATFORM_PATH)/$(PLATFORM_KEY)` is on the VPATH so its not really an issue.

tzarc, 2022-09-30:
Probably worth making it explicit, though. Will fix.

closed 2022-09-30 by drashna