qmk/qmk_firmware #18531: [Bug] Pressing a symbol before unpress another one using LOWER or RAISE layer

danyelvarejao, 2022-09-29:
## Describe the Bug

The bug occurs when all the symbols that need the SHIFT pressed. Like: []()!@$
What I think that is happen is that the SHIFT (auto) is getting unpressed before the right time.

How to do:
Press  the key LOWER or RAISE to change the layout that contains the key that is responsable for the symbol
Press and hold the symbol key
Press another symbol and hold it
Release the key that was getting pressed before

PS: To happen the bug, it's necessary to do this task real quick, as I show on the GIF's below

### Expected behaviour
[image]

### Actual behaviour
[image]

## System Information

**Keyboard: Planck EZ Glow ** 
**Revision (if applicable):** 
**Operating system: All, but i'm using Windows 11** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Danyel/Desktop/qmk_firmware
Ψ Detected Windows 10 (10.0.22621).
Ψ Git branch: master
Ψ Repo version: 0.18.8
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-09-29 07:31:40 -0300 (f871989f9c) -- add comments to layers and change variables name
Ψ - Latest upstream/master: 2022-09-29 01:32:25 -0700 (eb55551e2d) -- Monoflex 60 Layout Refactor and Touch-Up (#18517)
Ψ - Latest upstream/develop: 2022-09-29 08:33:39 +0000 (0e01759c61) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2022-09-29 11:02:36 +1000 (c71bc77da1) -- Bump anothrNick/github-tag-action from 1.46.0 to 1.49.0 (#18515)
Ψ - Common ancestor with upstream/develop: 2022-09-29 11:02:36 +1000 (c71bc77da1) -- Bump anothrNick/github-tag-action from 1.46.0 to 1.49.0 (#18515)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

mlcivilengineer, 2022-10-07:
Can confirm, it is happening to me as well on a moonlander keyboard both in Ubuntu 22.04 and Windows 11

27medkamal, 2022-10-10:
I can confirm this too on a moonlander and macos

mctaylorpants, 2022-11-19:
I'm seeing this issue as well on an ErgoDox EZ. Here's my QMK file and a GIF - in my case, I have a secondary layer with `=` but it outputs `+` when the keystroke is preceded by another modifier.

```
#include QMK_KEYBOARD_H
#include "version.h"
#include "keymap_german.h"
#include "keymap_nordic.h"
#include "keymap_french.h"
#include "keymap_spanish.h"
#include "keymap_hungarian.h"
#include "keymap_swedish.h"
#include "keymap_br_abnt2.h"
#include "keymap_canadian_multilingual.h"
#include "keymap_german_ch.h"
#include "keymap_jp.h"
#include "keymap_korean.h"
#include "keymap_bepo.h"
#include "keymap_italian.h"
#include "keymap_slovenian.h"
#include "keymap_lithuanian_azerty.h"
#include "keymap_danish.h"
#include "keymap_norwegian.h"
#include "keymap_portuguese.h"
#include "keymap_contributions.h"
#include "keymap_czech.h"
#include "keymap_romanian.h"
#include "keymap_russian.h"
#include "keymap_uk.h"
#include "keymap_estonian.h"
#include "keymap_belgian.h"
#include "keymap_us_international.h"
#include "keymap_croatian.h"
#include "keymap_turkish_q.h"
#include "keymap_slovak.h"

#define KC_MAC_UNDO LGUI(KC_Z)
#define KC_MAC_CUT LGUI(KC_X)
#define KC_MAC_COPY LGUI(KC_C)
#define KC_MAC_PASTE LGUI(KC_V)
#define KC_PC_UNDO LCTL(KC_Z)
#define KC_PC_CUT LCTL(KC_X)
#define KC_PC_COPY LCTL(KC_C)
#define KC_PC_PASTE LCTL(KC_V)
#define ES_LESS_MAC KC_GRAVE
#define ES_GRTR_MAC LSFT(KC_GRAVE)
#define ES_BSLS_MAC ALGR(KC_6)
#define NO_PIPE_ALT KC_GRAVE
#define NO_BSLS_ALT KC_EQUAL
#define LSA_T(kc) MT(MOD_LSFT | MOD_LALT, kc)
#define BP_NDSH_MAC ALGR(KC_8)
#define SE_SECT_MAC ALGR(KC_6)

enum custom_keycodes {
  RGB_SLD = EZ_SAFE_RANGE,
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
  [0] = LAYOUT_ergodox_pretty(
    KC_TRANSPARENT, KC_1,           KC_2,           KC_3,           KC_4,           KC_5,           KC_TRANSPARENT,                                 KC_TRANSPARENT, KC_6,           KC_7,           KC_8,           KC_9,           KC_0,           KC_BSPACE,
    KC_TAB,         KC_Q,           KC_W,           KC_E,           KC_R,           KC_T,           KC_TRANSPARENT,                                 KC_TRANSPARENT, KC_Y,           KC_U,           KC_I,           KC_O,           KC_P,           KC_TRANSPARENT,
    KC_ESCAPE,      KC_A,           KC_S,           KC_D,           KC_F,           KC_G,                                                                           KC_H,           KC_J,           KC_K,           KC_L,           KC_SCOLON,      KC_QUOTE,
    KC_LSHIFT,      KC_Z,           KC_X,           KC_C,           KC_V,           KC_B,           KC_LBRACKET,                                    KC_RBRACKET,    KC_N,           KC_M,           KC_COMMA,       KC_DOT,         KC_SLASH,       KC_RSHIFT,
    KC_LCTRL,       KC_TRANSPARENT, KC_LALT,        KC_LGUI,        MO(1),                                                                                                          KC_RGUI,        MO(2),          KC_TRANSPARENT, KC_TRANSPARENT, LM(3,MOD_LSFT),
                                                                                                    KC_TRANSPARENT, MO(1),          KC_LEFT,        KC_RIGHT,
                                                                                                                    KC_TRANSPARENT, KC_UP,
                                                                                    KC_LALT,        KC_ENTER,       MO(2),          KC_DOWN,        KC_TRANSPARENT, KC_SPACE
  ),
  [1] = LAYOUT_ergodox_pretty(
    KC_TRANSPARENT, KC_F1,          KC_F2,          KC_F3,          KC_F4,          KC_F5,          KC_TRANSPARENT,                                 KC_TRANSPARENT, KC_F6,          KC_F7,          KC_F8,          KC_F9,          KC_F10,         KC_TRANSPARENT,
    KC_TRANSPARENT, KC_EXLM,        KC_AT,          KC_LPRN,        KC_RPRN,        KC_PIPE,        KC_TRANSPARENT,                                 KC_TRANSPARENT, KC_NO,          KC_NO,          KC_MINUS,       KC_NO,          KC_ASTR,        KC_TRANSPARENT,
[... 55 lines cut ...]
    case 5:
      ergodox_right_led_1_on();
      ergodox_right_led_3_on();
      break;
    case 6:
      ergodox_right_led_2_on();
      ergodox_right_led_3_on();
      break;
    case 7:
      ergodox_right_led_1_on();
      ergodox_right_led_2_on();
      ergodox_right_led_3_on();
      break;
    default:
      break;
  }
  return state;
};
```

**Expected: while holding down Layer 1 and hitting K followed by J, I should get `_=`:**
[image]

**Actual: while holding down Layer 1 and hitting K followed by J in quick succession, I sometimes get `_+`:
[image]

drashna, 2022-12-10:
This may be fixed.  Please update the repo to 0.19.x and test again

danyelvarejao, 2022-12-10:
> This may be fixed. Please update the repo to 0.19.x and test again

I did the tests and it wasn't fixed.
I used the latest version to test.