qmk/qmk_firmware #18547: [Bug] Anne Pro 2 Bluetooth Not Working

y-mx-b, 2022-09-30:
## Describe the Bug

After flashing the QMK firmware, Bluetooth doesn't work at all. Using the Bluetooth keys to switch between connections doesn't work (though the LEDs will flash), and holding them down to get to pairing mode doesn't do anything either.

## System Information

**Keyboard:** Anne Pro 2
**Revision (if applicable):** C18
**Operating system:** macOS
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/user/qmk_firmware
Ψ Detected macOS 12.6 (Apple Silicon).
Ψ Git branch: master
Ψ Repo version: 0.18.8
Ψ - Latest master: 2022-09-30 15:13:21 +0900 (10005e9f39) -- Rotary encoder setting is revised to solve a malfunction. (#18227)
Ψ - Latest upstream/master: 2022-09-30 15:13:21 +0900 (10005e9f39) -- Rotary encoder setting is revised to solve a malfunction. (#18227)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-09-30 15:13:21 +0900 (10005e9f39) -- Rotary encoder setting is revised to solve a malfunction. (#18227)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

Everything works fine while the keyboard is wired with USB-C and the [Anne Pro 2 Shine](https://github.com/OpenAnnePro/AnnePro2-Shine) LED controller firmware works as well. It's just the Bluetooth functionality that isn't working.

tzarc, 2022-09-30:
Can confirm. No fix known as yet.

Jpe230, 2022-09-30:
After the latest update from Obins it seems that the Bluetooth protocol was changed.

Pre-update everything worked fine, after updating the BT module and flashing QMK my Bluetooth stopped working.

y-mx-b, 2022-10-01:
Would it be possible to flash an older Bluetooth firmware on the device then?

Asion001, 2022-10-11:
> Would it be possible to flash an older Bluetooth firmware on the device then?
@y-mx-b 
Yes. You can choose in obins archive version. I try couple latest's, and only version before 'Update to Bluetooth 5.0' work. After flashing from obins, flash shine and qmk.

SgoSkz, 2022-11-03:
#17483 might be what you're looking for. Sometimes I need to press the BT key twice before it actually connects. If you haven't tried out the fix, I'd recommend trying it out.

tzarc, 2022-11-04:
Tried flashing BLE firmwares 2.03, 2.04, and 2.05 onto my C18, none want to cooperate.

sehrgut, 2022-12-12:
Now that #17483 has been merged, does this bug still exist?

tzarc, 2022-12-12:
Yes.

dev-jelo, 2023-03-31:
> Tried flashing BLE firmwares 2.03, 2.04, and 2.05 onto my C18, none want to cooperate.

I also have 2.05 BLE firmware on a C18. I just flashed QMK and shine onto my keyboard and I thought the bluetooth wasn't working either, but then I tried pressing the bluetooth key twice (as suggested by @SgoSkz) and it worked. I'm on linux though so not sure if that makes any difference.

tzarc, 2024-08-09:
Had a stab at flashing a super old version of the BLE firmware, and this time QMK actually pairs and types correctly.

https://releases.obins.net/annepro2c18/firmware/1.14.1-alpha/ap2_c18_22_ble.bin worked for my C18 and current `develop` branch of QMK.

tzarc, 2024-08-09:
Full list of C18 BLE firmwares available:

https://releases.obins.net/annepro2c18/firmware/1.12.7-alpha/ble-c18-1.00-alpha.bin
https://releases.obins.net/annepro2c18/firmware/1.12.8/ble-c18-1.00.bin
https://releases.obins.net/annepro2c18/firmware/1.12.9-alpha/ble-c18-1.00.bin
https://releases.obins.net/annepro2c18/firmware/1.13.1-alpha/ble-c18-2.00-alpha.bin
https://releases.obins.net/annepro2c18/firmware/1.13.2-alpha/ble-c18-2.00-alpha.bin
https://releases.obins.net/annepro2c18/firmware/1.14.1-alpha/ap2_c18_22_ble.bin
https://releases.obins.net/annepro2c18/firmware/1.14.2-alpha/ap2_c18_22_ble.bin
https://releases.obins.net/annepro2c18/firmware/1.14.3-alpha/ap2_c18_22_ble.bin
https://releases.obins.net/annepro2c18/firmware/1.14.5-alpha/ap2_c18_23_ble.bin
https://releases.obins.net/annepro2c18/firmware/1.14.6/ap2_c18_23_ble.bin
https://releases.obins.net/annepro2c18/firmware/1.14.7-alpha/ap2_c18_25_ble.bin
https://releases.obins.net/annepro2c18/firmware/1.15.1-alpha/ap2_c18_25_ble.bin
https://releases.obins.net/annepro2c18/firmware/2.34.6/ap2_c18_0205_ble.bin
https://releases.obins.net/annepro2c18/firmware/2.36.3/ap2_c18_0205_ble.bin

Some of them have the same filename but are actually different firmware, so even if you try `ap2_c18_22_ble.bin` from `1.14.1-alpha`, it's actually different in `1.14.2-alpha`. You may need to try more than one to find a working copy.

tzarc, 2024-08-11:
Full list of C15 BLE firmwares:

https://releases.obins.net/annepro2/firmware/1.12.7-alpha/ble-c15-1.00-alpha.bin
https://releases.obins.net/annepro2/firmware/1.12.8/ble-c15-1.00.bin
https://releases.obins.net/annepro2/firmware/1.12.9-alpha/ble-c15-1.00.bin
https://releases.obins.net/annepro2/firmware/1.13.2-alpha/ble-c15-2.00-alpha.bin
https://releases.obins.net/annepro2/firmware/1.14.3/ap2_c15_22_ble.bin
https://releases.obins.net/annepro2/firmware/1.14.6/ap2_c15_23_ble.bin

SgoSkz, 2025-01-23:
The issue https://github.com/qmk/qmk_firmware/issues/23900 is related to lighting. However, both the lighting and bluetooth are controlled via UART. After changing the lighting UART, bluetooth functionality has been far more stable. Could someone else try it out to make sure it isn't a special case for my board? Thanks

Here are the exact changes mentioned
> https://github.com/OpenAnnePro/AnnePro2-Shine/blob/master/main.c#L29
> https://github.com/qmk/qmk_firmware/blob/master/keyboards/annepro2/annepro2.c#L32
> 
> For anyone still stuck on this error, changing these two lines from 115200 to 57600 seems to fix this issue for me.

If you don't change the UART on Shine, RGB lighting will not function
firmware from official qmk repo
keymap: default
board: C18
No shine or qmk modifications aside from those listed above

mkopec, 2025-03-06:
Does anyone know which BLE firmware versions work best with current qmk? I can't get the Caps Lock LED to work over BT, but it works over USB.