qmk/qmk_firmware #18568: [Bug] USB detection is inconsistent between config.h and info.json

TweetyDaBird, 2022-10-02:
## Describe the Bug

Adding the 'same' values in info.json as in config.h, has inconsistent behavior. With this in info.json the keyboard randomly times out upon insertion (4 times out of 25 in a row), and simply doesn't work at all, while it works flawlessly in config.h and has done so for months without a single timeout. Raising the timeout value to up to 50 made no noticeable difference (5 out of 10).

info.json		
```
"usb_detect": { 
			"enabled": true,
			"polling_interval": 2000,
			"timeout": 10
		},
```
config.h
```
#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT 2000
#define SPLIT_USB_TIMEOUT_POLL 10
```

## System Information

**Keyboard:** Lotus 58 Glow
**Revision (if applicable):** 
**Operating system:** Windows 10
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: F:/users/main/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: Lotus58Glow
Ψ Repo version: 0.18.8
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest Lotus58Glow: 2022-10-02 23:21:59 +0200 (1d5f8be664) -- Requested changes
Ψ - Latest upstream/master: 2022-10-03 07:24:22 +1100 (9ecd6eb9b9) -- Add `tzarc/ghoul`. (#18562)
Ψ - Latest upstream/develop: 2022-10-02 20:24:56 +0000 (bdffe9efe8) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2022-09-30 13:45:21 -0700 (dfa5880144) -- Smith and Rune Iron 165R2 Refactor (#18552)
Ψ - Common ancestor with upstream/develop: 2022-09-30 13:45:21 -0700 (dfa5880144) -- Smith and Rune Iron 165R2 Refactor (#18552)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

daskygit, 2022-10-06:
You have poll and timeout swapped in info.json, to match config.h it would be
```json
  "split": {
    "usb_detect": {
        "enabled": true,
        "timeout": 2000,
        "polling_interval": 10
    }
  },
 ```

daskygit, 2022-10-06:
You're welcome, closing the issue.

closed 2022-10-06 by daskygit

TweetyDaBird, 2022-10-07:
Actually, can we re-open this? (Or should I open a new issue?)

I figured out my problem wasn't entirely self inflicted after all. It seems there is a bug is in the conversion script to .json that is responsible for the switched variables.