qmk/qmk_firmware #18571: [Bug] The slave side hangs up at cold start of PC in the case of a split keyboard with Raspberry Pi Pico

yoichiro, 2022-10-03:
The slave side certainly hangs up at cold start of a PC in the case of a split keyboard with Raspberry Pi Pico.

## Describe the Bug

I have created a [Lunakey Pico](https://github.com/qmk/qmk_firmware/tree/master/keyboards/yoichiro/lunakey_pico) keyboard which is the split keyboard and each side has a Raspberry Pi Pico. Each side is connected by TRRS cable. When a cold startup of my PC, the slave side certainly hangs up. Then, when reconnecting the USB cable, it works.

The following list is I tried:

* The circuit of the left side has GP22 which is connected to VCC. The circuit of the right side has GP22 which is connected to GND. I tried to specify the `#define SPLIT_HAND_PIN GP22`, nothing happened.
* I know that the GP24 pin becomes HIGH at connecting USB cable. I tried to specify the `#define USB_VBUS_PIN GP24`, but nothing happened.

## System Information

**Keyboard:** [Lunakey Pico](https://github.com/qmk/qmk_firmware/tree/master/keyboards/yoichiro/lunakey_pico)
**Operating system:** Microsoft Windows 11 and Ubuntu Linux 22.04
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/yoichiro/projects/keyboard/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/yoichiro/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.18.8
Ψ - Latest master: 2022-10-01 12:54:42 +1000 (9f0d9b4fbe) -- onekey: fix quine keymap (#18555)
Ψ - Latest upstream/master: 2022-10-03 12:26:04 +1100 (cc1565f2a3) -- `:flash`: print bootloader (#18569)
Ψ - Latest upstream/develop: 2022-10-02 09:36:07 +1100 (c4ff8192c2) -- Reverse keymap search order (#18449)
Ψ - Common ancestor with upstream/master: 2022-10-01 12:54:42 +1000 (9f0d9b4fbe) -- onekey: fix quine keymap (#18555)
Ψ - Common ancestor with upstream/develop: 2022-10-01 12:54:42 +1000 (9f0d9b4fbe) -- onekey: fix quine keymap (#18555)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

tzarc, 2022-10-03:
Possibly related to #18511?

yoichiro, 2022-10-03:
@tzarc I guess that this issue is not same as the #18511, because the master side certaily works normally but the slave side does not work in my case.

drashna, 2022-10-04:
Might be related to https://github.com/qmk/qmk_firmware/pull/18574

yoichiro, 2022-10-04:
@drashna I will try latest code set to confirm whether this issue is reproduced or not later.

yoichiro, 2022-10-08:
@drashna Unfortunately, this issue still occurs using the latest develop branch. Both sides certainly hang up at cold starting of my PC.

KarlK90, 2022-10-08:
Did you update the `chibios-contrib` submodule when switching to the `develop` branch? There is a critical bug fix for the USB driver. The easiest way to update would be to run `make git-submodule` in the root folder.

yoichiro, 2022-10-10:
@KarlK90 I have already updated all sub modules and tried when switching to the `develop` branch. However, this issue was reproduced.

Then, I tried using `SPLIT_USB_TIMEOUT 5000` and `SPLIT_USB_TIMEOUT_POLL 25`, and I was able to confirm that my keyboard was normally recognized and works. I close this issue. Thanks @drashna @KarlK90 @tzarc for advices!

closed 2022-10-10 by yoichiro

yoichiro, 2022-10-11:
@drashna @tzarc @KarlK90 I have tested the latest develop branch and the `#define USB_VBUS_PIN GP24` and the `#define SPLIT_HAND_PIN GP22` for Lunakey Pico which is a split keyboard using Raspberry Pi Pico just now. As the result, the keyboard finally normally works without using the `SPLIT_USB_DETECT`.

The slave side of the keyboard gets VCC from VSYS pin, therefore I thought that the `USB_VBUS_PIN` is available.

[image]

Raspberry Pi Pico has the GPIO24 pin to detect whether VBUS is provided from USB or not.

[image]

Using the develop branch, my Lunakey Pico keyboard launches normally and works without any delay of the `SPLIT_USB_DETECT` at cold starting of my PC.