qmk/qmk_firmware #18591: [Bug] kb2040 keyboard unresponsive after resuming from Windows hibernate

molohov, 2022-10-04:
### Describe the Bug

After flashing the fixes from https://github.com/qmk/qmk_firmware/issues/18511 onto my board, Windows no longer throws "USB device malfunctioned" errors after resuming from hibernate, which would normally require me to reconnect the USB cable to get the keyboard working again.

However, after the fix, the keyboard is still unresponsive after resuming. If I reset the MCU, the keyboard will be recognized by Windows. 

My keyboard is connected to my computer through a Thunderbolt dock, which may complicate the problem.

### Keyboard Used

ursula

### Operating System

Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.       
Ψ CLI version: 1.1.0
Ψ QMK home: (REDACTED)
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: rp2040
Ψ Repo version: 0.18.9
Ψ - Latest rp2040: 2022-10-03 15:27:30 -0700 (d39acc0e15) -- Changes for latest develop + rp2040, hopefully this resolves the bootup issue
Ψ - Latest upstream/master: 2022-10-03 09:41:24 +0200 (1d194ac60f) -- Adds Anne Pro 2 c18 ISO variant support (#18154)
Ψ - Latest upstream/develop: 2022-10-03 19:57:13 +0200 (976f454df0) -- [Bug] Update ChibiOS-Contrib for USB IRQ and bus handling fixes (#18574)
Ψ - Common ancestor with upstream/master: 2022-10-03 09:41:24 +0200 (1d194ac60f) -- Adds Anne Pro 2 c18 ISO variant support (#18154)
Ψ - Common ancestor with upstream/develop: 2022-10-03 19:57:13 +0200 (976f454df0) -- [Bug] Update ChibiOS-Contrib 
for USB IRQ and bus handling fixes (#18574)
Ψ All dependencies are installed.       
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2    
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)  
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)

Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)  
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)    
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2022-12-10:
is this still happening on 0.19.x?

molohov, 2022-12-31:
Sorry for the late response. I just rebased my tree on latest develop (724aa3a4b256ccb058e4ef86c882167635d6815d) and the issue hasn't occurred on today's startup. I will test for a few more days.

molohov, 2023-01-02:
Unfortunately this issue is still persisting as of the above commit.

KarlK90, 2023-01-21:
@molohov if you have RGB leds on your keyboard, could you please check if my fixes to the ws2812 vendor driver in https://github.com/qmk/qmk_firmware/pull/19652 fix your deadlocks?

molohov, 2023-01-25:
My board does not have RGBs soldered or enabled: https://github.com/molohov/qmk_firmware/tree/rp2040/keyboards/ursula/keymaps/molohov