qmk/qmk_firmware #18627: [Bug] shifted character outputting non shifted version on Linux when typing really fast

mlcivilengineer, 2022-10-07:
### Describe the Bug

I have 2 layers, one 0 layer that outputs "d" and "f", and if I hold the modifier it goes to layer 1 and outputs "(" and ")".

The problem is if I hold the modifier and start typing really fast ()()()()()()()(). The output that I get is: ()()()(09)()()()()()90909090909)()()()90()()90() instead of ()()()()()()()()()()()()()()()()()()()()()()()()()()()()()()()(). It seems I get the unshifted version of the braces out of nowhere and I don't understand why this is happening. This also happen with other keys such as _ outputting - instead. This problem is occurring on Ubuntu 22.04 and the bug does not appear in another keyboard (corsair k68)  so I know the problem is not in the computer itself. I tried using the moonlander on windows 11 on my machine and the bug does not occur there. So right now the problem is in the interaction between Linux Ubuntu 22.04 and the firmware inside Moonlander.

The following link shows my current layout: https://configure.zsa.io/moonlander/layouts/0LOEP/QEYlV/0

### Keyboard Used

Moonlander

### Link to product page (if applicable)

https://www.zsa.io/moonlander/

### Operating System

Ubuntu 22.04

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

my keymap.c file:
``` C
#include QMK_KEYBOARD_H
#include "version.h"
#include "keymap_german.h"
#include "keymap_nordic.h"
#include "keymap_french.h"
#include "keymap_spanish.h"
#include "keymap_hungarian.h"
#include "keymap_swedish.h"
#include "keymap_br_abnt2.h"
#include "keymap_canadian_multilingual.h"
#include "keymap_german_ch.h"
#include "keymap_jp.h"
#include "keymap_korean.h"
#include "keymap_bepo.h"
#include "keymap_italian.h"
#include "keymap_slovenian.h"
#include "keymap_lithuanian_azerty.h"
#include "keymap_danish.h"
#include "keymap_norwegian.h"
#include "keymap_portuguese.h"
#include "keymap_contributions.h"
#include "keymap_czech.h"
#include "keymap_romanian.h"
#include "keymap_russian.h"
#include "keymap_uk.h"
#include "keymap_estonian.h"
#include "keymap_belgian.h"
#include "keymap_us_international.h"
#include "keymap_croatian.h"
#include "keymap_turkish_q.h"
#include "keymap_slovak.h"

#define KC_MAC_UNDO LGUI(KC_Z)
#define KC_MAC_CUT LGUI(KC_X)
#define KC_MAC_COPY LGUI(KC_C)
#define KC_MAC_PASTE LGUI(KC_V)
#define KC_PC_UNDO LCTL(KC_Z)
#define KC_PC_CUT LCTL(KC_X)
#define KC_PC_COPY LCTL(KC_C)
#define KC_PC_PASTE LCTL(KC_V)
#define ES_LESS_MAC KC_GRAVE
[... 255 lines cut ...]
    dance_state[1].step = dance_step(state);
    switch (dance_state[1].step) {
        case SINGLE_TAP: register_code16(KC_UP); break;
        case DOUBLE_TAP: register_code16(KC_UP); register_code16(KC_UP); break;
        case DOUBLE_HOLD: register_code16(KC_PGUP); break;
        case DOUBLE_SINGLE_TAP: tap_code16(KC_UP); register_code16(KC_UP);
    }
}

void dance_1_reset(qk_tap_dance_state_t *state, void *user_data) {
    wait_ms(10);
    switch (dance_state[1].step) {
        case SINGLE_TAP: unregister_code16(KC_UP); break;
        case DOUBLE_TAP: unregister_code16(KC_UP); break;
        case DOUBLE_HOLD: unregister_code16(KC_PGUP); break;
        case DOUBLE_SINGLE_TAP: unregister_code16(KC_UP); break;
    }
    dance_state[1].step = 0;
}

qk_tap_dance_action_t tap_dance_actions[] = {
        [DANCE_0] = ACTION_TAP_DANCE_FN_ADVANCED(on_dance_0, dance_0_finished, dance_0_reset),
        [DANCE_1] = ACTION_TAP_DANCE_FN_ADVANCED(on_dance_1, dance_1_finished, dance_1_reset),
};
```

mlcivilengineer, 2022-10-07:
reading through the issues, I think it is the same problem happening on the issue #18531

drashna, 2022-12-10:
This is most likely 17569

OscarCreator, 2023-03-12:
I can confirm that this is still an issue on Ubuntu 22.04. And on Windows 10 there is no problem.
Any updates if this will get fixed soon? As it a very annoying bug.

mlcivilengineer, 2023-03-12:
> I can confirm that this is still an issue on Ubuntu 22.04. And on Windows 10 there is no problem. Any updates if this will get fixed soon? As it a very annoying bug.

With my experiments, the problem was OS agnostic. You may not be experiencing the issue on Windows because the repeat delay of the keyboard presses are set to a high value. Try setting the repeat delay to a much lower value and the issue may appear, at least that is what I found.

OscarCreator, 2023-03-16:
> With my experiments, the problem was OS agnostic. You may not be experiencing the issue on Windows because the repeat delay of the keyboard presses are set to a high value. Try setting the repeat delay to a much lower value and the issue may appear, at least that is what I found.
I tried adjusting all the setting related to keyboard on Ubuntu but it did not work. Any other way to fix this?

OscarCreator, 2023-03-19:
Not 100% sure we had the same issue but I solved my problem by changing my layer key from `TT(1)` to `MO(1)`.