qmk/qmk_firmware #18636: [Feature Request] Add Drop SHIFT support

purist69, 2022-10-07:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Apologies if this has been requested previously, but the Drop Configurator is limited in functionality for the Drop SHIFT keyboard, if would be great if it was supported with QMK directly to allow for additional macro functionality.

adophoxia, 2022-10-07:
As far as I've found for the SHIFT, Drop actually has their own [fork of QMK](https://github.com/Massdrop/qmk_firmware/tree/master/keyboards/massdrop/shift) which actually has the SHIFT keyboard and a few others besides the Ctrl and Alt. It is a yr old, so if you can merge via upstream successfully, you may have a chance of being able to compile the firmware.

purist69, 2022-10-07:
Unfortunately thats exactly why I'm asking.  The Massdrop QMK fork version of the SHIFT profile is two years old.  QMK has matured beyond the profile MD has publicly listed.  When I contacted Drop support directly all I got, litterally, was:

"The link that was provided to you is Drop's own QMK Configurator. We do not have a downloadable version. You will need to use that and go through the Shift configurator within it.
Unfortunately, that is all of the information we have and are able to provide to you."

So ya, no love from Drop.

zvecr, 2022-10-07:
> The Massdrop QMK fork version of the SHIFT profile is two years old

You just need to know where to look <https://github.com/Massdrop/qmk_firmware/tree/feature/riot_xap/keyboards/drop/shift>.

Drop have been deploying this as the "modern" option from their configurator and working through the various issues that have been raised from the move to RIOT. At this point its fairly mature, and the aim is to merge the work in the near future.

purist69, 2022-10-07:
Thank you for the link and info!!

closed 2022-10-07 by purist69

purist69, 2022-10-08:
For anyone interested, I was able to get a working configuration by including the KB profile within the QMK official version, however I needed to search all files, by name and content, you need to replace any files which include both RIOT and SAMD51.  I also needed to download the RIOT source and put it in the lib/riot folder.  The firmware compiled just fine, and worked.  But, via/vial do not compile successfully.