qmk/qmk_firmware #18644: [Bug][Linux] pico-sdk compile error on 1upkeyboards pi40 and pi60_hse

JohnCiubuc, 2022-10-08:
### Describe the Bug

I am trying to compile for 1upkeyboards pi40 keyboard. I am currently configuring it with VIA, but would rather use QMK (as I have several boards running QMK currently)

This is has been reproduced on two computers, both using fresh qmk_firmware repositories

when I run `make 1upkeyboards/pi40:default`, I get the following compile error:

```
QMK Firmware 0.18.13
Making 1upkeyboards/pi40 with keymap default

arm-none-eabi-gcc (Arch Repository) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_1upkeyboards_pi40/src/info_config.h                                          [OK]
Generating: .build/obj_1upkeyboards_pi40/src/default_keyboard.c                                     [OK]
Generating: .build/obj_1upkeyboards_pi40/src/default_keyboard.h                                     [OK]
Generating: .build/obj_1upkeyboards_pi40/src/layouts.h                                              [OK]
Compiling: keyboards/1upkeyboards/pi40/pi40.c                                                       [OK]
Compiling: .build/obj_1upkeyboards_pi40/src/default_keyboard.c                                      [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/bootmagic_lite.c                                                       [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: lib/printf/src/printf/printf.c                                                           [OK]
Compiling: quantum/mousekey.c                                                                       [OK]
Compiling: drivers/eeprom/eeprom_driver.c                                                           [OK]
Compiling: drivers/eeprom/eeprom_wear_leveling.c                                                    [OK]
Compiling: quantum/wear_leveling/wear_leveling.c                                                    [OK]
Compiling: platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash.c                    In file included from platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash.c:9:
In function 'rom_func_lookup_inline',
    inlined from 'pico_program_bulk' at platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash.c:129:91:
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:129:66: note: in expansion of macro 'rom_hword_as_ptr'
  129 |     rom_table_lookup_fn rom_table_lookup = (rom_table_lookup_fn) rom_hword_as_ptr(0x18);
      |                                                                  ^~~~~~~~~~~~~~~~
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:130:41: note: in expansion of macro 'rom_hword_as_ptr'
  130 |     uint16_t *func_table = (uint16_t *) rom_hword_as_ptr(0x14);
      |                                         ^~~~~~~~~~~~~~~~
In function 'rom_func_lookup_inline',
    inlined from 'pico_program_bulk' at platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash.c:130:83:
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
[... 23 lines cut ...]
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:359: .build/obj_1upkeyboards_pi40_default/wear_leveling_rp2040_flash.o] Error 1
Make finished with errors
make: *** [Makefile:414: 1upkeyboards/pi40:default] Error 1

```

It seems to be an issue with the pico sdk? I tried to make the default layout with the pi60_hse and I achieve the same error. Does anyone have any suggestions on what I should do going forward? 

### Keyboard Used

1upkeyboards/pi40

### Operating System

Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2022-10-08:
Can you confirm that this is still an issue on the `develop` branch?

JohnCiubuc, 2022-10-08:
This issue is resolved on the `develop` branch. `pi60_hse` compiles without any issues. `pi40` is unable to compile but this is due to a LAYOUT issue. I fixed the LAYOUT issue locally, and will submit a pull request for the fix.

Thanks for your help!

zvecr, 2022-10-09:
Fixed by pico-sdk update.

closed 2022-10-09 by zvecr