qmk/qmk_firmware #18648: [Bug] KC_GESC does not play nicely with Firefox

mkrasnitski, 2022-10-08:
### Describe the Bug

Trying to type backtick in in Firefox using `KC_GESC` does not work as intended. No backticks are inserted. Likely because if `KC_GESC` appears in the keymap, then the `Super` key is changed to only send its corresponding keycode on KeyUp. Now, when pressing `Super+Esc`, the backtick keycode is actually detected, but the text character is not inserted. Waiting for keyboard repeat to first kick in, then releasing `Super` while still holding `Esc`, allows for backticks to actually be inserted. A workaround I found is to map Fn+Esc to `KC_GRV`.

I understand this is not entirely a QMK-only bug, but I wanted to ask here to see if there are any other workarounds, or if anyone else has run into this issue before.

### Keyboard Used

dztech/dz60rgb_ansi/v1

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /media/mass/Keyboards/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.18.13
Ψ - Latest master: 2022-10-08 15:52:38 +1100 (7b3ad56f13) -- Fix future naming collisions. (#18645)
Ψ - Latest upstream/master: 2022-10-08 15:52:38 +1100 (7b3ad56f13) -- Fix future naming collisions. (#18645)
Ψ - Latest upstream/develop: 2022-10-08 11:13:58 -0500 (86938fd958) -- Allow QK_MAKE to work with converters (#18637)
Ψ - Common ancestor with upstream/master: 2022-10-08 15:52:38 +1100 (7b3ad56f13) -- Fix future naming collisions. (#18645)
Ψ - Common ancestor with upstream/develop: 2022-10-08 15:52:38 +1100 (7b3ad56f13) -- Fix future naming collisions. (#18645)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Running on Firefox 105.

jumski, 2023-10-19:
I have the same problem and confirm the repeat-release trick works.

Thanks for reporting this @mkrasnitski - the repeat-release is already a godsend, will test the KC_GRV asap when at my Linux workstation.

mkrasnitski, 2023-10-19:
You can work around this issue by mapping backtick to something else. I personally use `Fn+\`, or you could use `Fn+Esc`.