qmk/qmk_firmware #18694: [Feature Request] Implement feature that allows tapping of both/multiple shift keys to become caps-lock

BrokenGale, 2022-10-13:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

(Note the following is written by a programming novice that knows next to nothing about qmk. I have flashed some boards and modified some pre-loaded .c files and that's it, so much of the implementation difficulties have been hand-waved/ignored because I genuinely don't know if it would be possible/practical. Disclaimer aside, here is my proposal)

I have a dream, a dream of a future w/o the dreaded unintended caps-lock press. A dream where caps-lock is relegated to a feature requiring 2 keys we already have on our keyboard freeing up precious space on our 40% and 60% boards.

Okay enough preamble.
Hopefully I convinced you that caps-lock should not be taking up an entire function on our layers or even tapdances and it should be implemented instead as a deliberate press of 2 keys that most keyboards have. 

Think about it, there really isn't any reason to hit both shift keys at the same time and this could be an opportunity to implement features into modifier keys that often gets repeated for no particular reason. This could be a gateway into different ways that we can add more keys w/o remembering the specific action we had to do to a specific key. Just tap/hold and press the two shift keys and badabing.

Like I said at the beginning, this is just the ramblings of a mad dreamer with a crazy dream, if this is not possible/practical shut it down, I take no offense.

tzarc, 2022-10-13:
That key combination is already in use by the `command` system: https://docs.qmk.fm/#/feature_command?id=usage

joukewitteveen, 2022-10-16:
I use and like this feature; both shifts together toggle caps lock. To take care of the `command` system issue mentioned by @tzarc, I've put
```C
#define IS_COMMAND() (get_mods() == MOD_MASK_CTRL)
```
in my `config.h`. This changes the the combination that triggers the `command` system to pressing both control keys.

While it would probably be possible to toggle caps lock from `process_record_user`, there is a far easier option for Linux users that configure their keyboard with `xkeyboard-config` (i.e. all typical graphical setups). Those users can simply set the `XkbOption` [shift:both_capslock](https://gitlab.freedesktop.org/xkeyboard-config/xkeyboard-config/-/blob/xkeyboard-config-2.37/rules/base.xml#L7855-7858). In Gnome, this option is available through Gnome Tweaks as a compatibility option in the additional keyboard layout options.

getreuer, 2022-11-28:
Something similar to what you asked for: with [Caps Word](https://docs.qmk.fm/#/feature_caps_word) (rather than Caps Lock), you can use Left Shift + Right Shift to activate Caps Word if you add to config.h:

```c
#define BOTH_SHIFTS_TURNS_ON_CAPS_WORD
```

and avoid the `command` conflict by redefining `IS_COMMAND()` as @joukewitteveen describes above.

closed 2023-03-30 by github-actions[bot]