qmk/qmk_firmware #18746: [Bug] Not able to compile lily58 with RP2040

gabssanto, 2022-10-16:
### Describe the Bug

Currently I'm unable to run the compiler for Lily58 with RP2040.
Using this command: `qmk compile -kb lily58 -km default`
Having this output: 
[image]
with this changes to the keyboard:
[image]

### Keyboard Used

lily58

### Link to product page (if applicable)

https://keebd.com/products/splinky-rp2040-controller-usb-type-c

### Operating System

MacOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/gabssanto/qmk_firmware
Ψ Detected macOS 12.6 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.18.13
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-10-16 15:58:42 +1100 (81988cc032) -- Slipped through the cracks. (#18742)
Ψ - Latest upstream/master: 2022-10-16 15:58:42 +1100 (81988cc032) -- Slipped through the cracks. (#18742)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-10-16 15:58:42 +1100 (81988cc032) -- Slipped through the cracks. (#18742)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2022-10-16:
Changing only the bootloader type is entirely the wrong direction.

<https://docs.qmk.fm/#/feature_converters>

Splinky isnt currently supported as a converter type, but it claims `pinout is compatible with the widely available [SparkFun RP2040](https://www.sparkfun.com/products/18288)`

You should revert your changes, and run
```
qmk compile -kb lily58 -km default -e CONVERT_TO=promicro_rp2040
```

gabssanto, 2022-10-16:
Hey @zvecr thanks so much! It worked for me, I'm new to this, so I'm not completely sure yet of everything, thanks again!!

closed 2022-10-16 by zvecr