qmk/qmk_firmware #18748: [Bug] Ergodox Infinity hangs on first key press

arBmind, 2022-10-16:
### Describe the Bug

I wanted to try the latest develop build for my keyboard.
Building and flashing the firmware works fine.

The keyboard seems to boot just fine. The first key down is also reported to the operating system via USB.
But the keyboard seems to hang immediately.

I tried default and input_club keymaps. Same behaviour.

### Keyboard Used

input_club/ergodox_inifinity

### Operating System

WSL2 / Ubuntu 22.04

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /mnt/wsl/veracrypt1/projects/ergodox/qmk
Ψ Detected Linux (WSL).
⚠ I/O performance on /mnt may be extremely slow.
Ψ Git branch: feature/arbtype2
Ψ Repo version: 0.18.13
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.2.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Building and flashing my last version from January 2022 works fine. But there are a ton of commits in between and the keyboard folder was moved.

Any ideas on how to debug or investigate this further are welcome.

jtojnar, 2022-10-21:
I have also encountered freezes in https://github.com/qmk/qmk_firmware/issues/18805, maybe try reverting https://github.com/qmk/qmk_firmware/commit/2078a56369af376e3275f02e21d48ab6cc39bc36?

sigprof, 2022-10-22:
This certainly looks like the breakage accidentally introduced in #18631; the underlying reason is that *some* USB LLDs (for the KINETIS and MIMXRT1062 chips) have bugs, and the “useless” USB callbacks which got removed during the cleanup were actually masking those bugs.

The workaround in #18811 should bring the board back into the working state; alternatively, you may try testing these fixes for `chibios-contrib`:

- https://github.com/ChibiOS/ChibiOS-Contrib/pull/342
  (this covers the MK20DX256 chip on Ergodox Infinity)
- https://github.com/ChibiOS/ChibiOS-Contrib/pull/343
  (this one does not really apply to the Ergodox Infinity hardware)

closed 2022-10-23 by zvecr