qmk/qmk_firmware #18759: Unable to recognize when the keyboard is plugged in, use

jeromeli002, 2022-10-18:
### Issue Description

Some computers are powered on with computers plugged in, and the keyboard cannot be used. You need to plug it in again to use it. Is this a hardware problem or a firmware problem

jeromeli002, 2022-10-18:
Is it not good to support USB 3.0? Many online searches said that it would be better to replace the USB 2.0 port. Is there any way to completely solve the code?

jeromeli002, 2022-10-18:
Isn't it good to support usb-hub? The keyboard cannot be used when the computer is turned on when it is plugged into the hub. It needs to be re-plugged and unplugged before it can be used. Is this a hardware problem or a firmware problem?

harishsr, 2022-11-15:
I'm running into this, too. My keyboard is plugged into a USB hub, which is connected to my M1 MacBook Pro. Every time the computer sleeps, the keyboard has to be plugged back in.

This is a recent activity that didn't happy about a month ago, so I wonder if this has something to do with Mac OS Ventura.

drashna, 2022-12-10:
If this is a continued issue, please re-open as a bug issue, with the relevant information

closed 2022-12-10 by drashna