qmk/qmk_firmware #18797: [Feature Request] Add Keychron Q4 and V series boards (V4, V3, V2, V1)

lanes10000, 2022-10-21:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I bought a Keychron V4 and I noticed that there was no QMK firmware for this on the repo (firmware jsons are listed on manufacturer's website), and as such VIA is not automatically recognizing the board and I have to manually load the json for the keyboard. I imagine this is also an issue for other V series boards, and I also noticed that the Q4 does not have an entry in the repo, so I thought I'd mention that, since upon review of that json, it looks almost identical to the V4 (not sure if they're the same PCB or not).

Thanks!

lokher, 2022-12-11:
> V series boards are Sonix controllers, IIRC.
No, V series boards is almost same to Q Series, same MCU

lokher, 2022-12-11:
> ### Feature Request Type
> * [x]  Core functionality
> * [ ]  Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
> * [ ]  Alteration (enhancement/optimization) of existing feature(s)
> * [ ]  New [...]

We raise 2 or 3 keyboards PRs each time, Q10 is there, so we will start from V1 of V series soon. If you need the source code, you can get it from `playground` branch of Keychron fork 
https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron

closed 2023-01-06 by zvecr