qmk/qmk_firmware #18813: [Feature Request] Option to disable shifted KC_MINS when caps word is enabled

chdorner, 2022-10-22:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I have set up separate key combos for `KC_MINS` and `KC_UNDS`, but because the caps word feature automatically applies shift to `KC_MINS` in addition to all alpha characters I can't actually type the minus/dash key when caps word is enabled.

Looking at the caps word implementation this seems to be easy enough:
```diff
diff --git a/quantum/process_keycode/process_caps_word.c b/quantum/process_keycode/process_caps_word.c
index 1b9583196d..19f77bac59 100644
--- a/quantum/process_keycode/process_caps_word.c
+++ b/quantum/process_keycode/process_caps_word.c
@@ -170,9 +170,13 @@ __attribute__((weak)) bool caps_word_press_user(uint16_t keycode) {
     switch (keycode) {
         // Keycodes that continue Caps Word, with shift applied.
         case KC_A ... KC_Z:
+            add_weak_mods(MOD_BIT(KC_LSFT)); // Apply shift to next key.
+            return true;
+#ifndef NO_CAPS_WORD_SHIFTED_MINS
         case KC_MINS:
             add_weak_mods(MOD_BIT(KC_LSFT)); // Apply shift to next key.
             return true;
+#endif // NO_CAPS_WORD_SHIFTED_MINS

         // Keycodes that continue Caps Word, without shifting.
         case KC_1 ... KC_0:
```

Happy to provide a fully-fledged PR if this is considered an acceptable enhancement!

getreuer, 2022-11-28:
Yes, it is possible to achieve this behavior through `caps_word_press_user()`. I'm in a similar situation as you with separate keys for `KC_MINS` and `KC_UNDS`, and do the following in my keymap.c ([code link](https://github.com/getreuer/qmk-keymap/blob/a1f62991a99b205b7d493dcc1fdea3946cbb575c/keymap.c#L234-L253)):

```c
bool caps_word_press_user(uint16_t keycode) {
  switch (keycode) {
    // Keycodes that continue Caps Word, with shift applied.
    case KC_A ... KC_Z:
      add_weak_mods(MOD_BIT(KC_LSFT));  // Apply shift to the next key.
      return true;

    // Keycodes that continue Caps Word, without shifting.
    case KC_1 ... KC_0:
    case KC_BSPC:
    case KC_DEL:
    // I have a dedicated underscore key, so no need to shift KC_MINS.
    case KC_MINS:
    case KC_UNDS:
      return true;

    default:
      return false;  // Deactivate Caps Word.
  }
}
```

drashna, 2022-12-10:
This is definitely a persona preference thing.  Eg, I find the default behavior to be mostly fine.  

The point of the `caps_word_press_user` function is explicitly so that users can fine tune the behavior to their preferences.

That said, I'm gonna close this, since it seems resolved.

closed 2022-12-10 by drashna